volumio/Volumio2 #706: NAS mount fails when setting a name containing a space

PierreMardon, 2016-09-18:
The drive doesn't show up in the list of drives, successfully mounting a drive with no space in the name makes the new one and the previous one appear.
But deleting the first (failing) one fails according to the red feedback I get, even if it's deleted from the list of network drives.
Don't know if it can help but the drive appears empty in my library with this name `FirstWord\ SecondWord`, and a directory is created under `/mnt/NAS/` with the name `FirstWord\\\ SecondWord`.
Seems like a tiny escaping problem ;)

(btw, Volumio is great :) )

volumio, 2016-09-28:
Fixed!

closed 2016-09-28 by volumio

PierreMardon, 2016-10-03:
Nice :)

EnerGeh, 2017-06-27:
Found another escaping problem in v2.201: a password containing 

> &

 will result in mounting error. When I set the whole password in quotes 

> "Pass&Word"

 it works.

(Btw: Volumio is awsome and made me to buy at least 4 DACs in your shop. I love it! :))