volumio/Volumio2 #713: Drilldown in search results

Atrejoe, 2016-09-21:
I really love the way the search bar gives unified results, allowing me to search across my local music as well as Spotify (or other streaming services in the future).

However when I get a search result that indicates a group (of tracks), like artist, album or genre, it would be very nice to select that group and drill down to the tracks contained in that group (or even the groups withing that group).

Like search for `Pink` one would get:

- Artists:
    - `Pink`
    - `Pink Floyd`
- Tracks:
    - `Pink is my favorite color` - Aerosmith
    - _etc._

It would be nice to be able to select a 'group' like `Pink Floyd` and see its albums (`Shine on you crazy diamond (Parts 1-5)`) and tracks (`Wish you were here`) and be able to browse, enqueue or play them.

This is similar to the library view, and of course the respective API's of streaming services should be able to support this.

volumio, 2016-09-24:
Done

closed 2016-09-24 by volumio

Atrejoe, 2016-09-26:
Wow this response time is fantastic, thanks a lot!
I'm looking forward to it!

volumio, 2016-09-26:
https://volumio.org/forum/volumio-final-release-very-near-pre-test-needed-t4709.html

Enjoy!