volumio/Volumio2 #714: 'mpc prev' just jumps to the start of the song

martin2250, 2016-09-21:
as the title states, the mpd emulation interface does not like the 'prev' command and just jumps to the start of the song instead of playing the previous song.

volumio, 2016-09-21:
The mpd emulation interface is not active. Would you help us finishing that?

martin2250, 2016-09-21:
if it isn't active, how is mpc communicating with volumio at all?
I had a quick look over the source code but all I could find were some relaying functions that couldn't trace any further (Volumio2/app/plugins/music_service/mpd/ and Volumio2/app/plugins/user_interface/mpdemulation/).
I could certainly try if you point me to the correct branch/file

volumio, 2016-09-21:
Volumio communicates directly with mpd, therefore mpc talks directly to mpd . 

To clear:
Volumio2/app/plugins/music_service/mpd/
is the mpd plugin, the one that sends commands from Volumio to MPD

Volumio2/app/plugins/user_interface/mpdemulation/
This is the mpd emulation plugin. Now it is inactive (to enable it edit the /data/configuration/plugins.json and change its status to enabled true) 

What the mpd emulation is supposed to do is mimic mpd protocol to allow mpd client to see volumio playlists, play queue, search (so to see also spotify items) etc

We can tackle this together, and mpd emulation would be really appreciated...

martin2250, 2016-09-21:
so the new 'prev' behavior comes from mpc itelf? it used to work on volumio 1.
what is the best way for restarting the node.js server? 'killall node' and waiting for the watchdog to restart the three instances doesn't seem to work too well (and console output would be pretty handy to have).

volumio, 2016-09-21:
Yes the mpc prev comes from mpc (probably newer versions of mpd handle this differently). 

The best way to restart is 

killall node 

and watch the log via 

sudo journalctl -f 

Alteratively, stop volumio service by 

systemctl stop volumio.service 

and start it manually by 

cd /volumio
node index.js

closed 2016-09-28 by volumio