volumio/Volumio2 #717: unsecured wifi access point

epegzz, 2016-09-23:
Hey :)

I realized that volumio provides a public access point. When connecting to it the player interface automatically opens up. So theoretically my neighbors could play music by just connecting to the open access point.

Is that on purpose? Is there any option set a password? Or to easily disable this feature?

volumio, 2016-09-23:
Not yet, but we'll be adding such feature soon

epegzz, 2016-09-23:
I'm living in Berlin city center at a very touristy place with lot's of bars, cafes and night life. So people try to see if there's free wifi 24/7 and when they connect to volumio's public access point then my player opens up to them letting them play music at any volume at any time in my flat…

A big fat warning about this feature should be on the website of volumio, because probably not many people are aware of the data and access they leak.

So, the only way to disable this feature is blacklisting the wifi kernel module I guess?

volumio, 2016-09-23:
You're right, I made a thread on how to fix this issue (waiting for us to fix it)

https://volumio.org/forum/important-how-secure-volumio-hotspot-t4705.html

closed 2016-09-23 by volumio

epegzz, 2016-09-23:
awesome, thx!