volumio/Volumio2 #726: albumart not loaded because of special characters

CaptainMetal, 2016-09-25:
i find something out about the albumart covers loaded from internet automatically:
the covers are not correctly loaded when there are special characters in the album title or artists name  (or may be the directory path of the musicfiles ?) like:

&
/
- 
!
[
]

this is v.0991 and happens in the browsetab viewing albums or artists and also in the playbacktab.
when the picturefile exists as file in the musicfile directory, then in the playbacktab the albumart is displayed correctly, but not on the browsetab.
and sometimes there is just an X. 

regards,
CaptainMetal 

[image]
[image]
[image]

xipmix, 2016-11-29:
It looks like this can be closed - @CaptainMetal can you try with 2.031 which should have the fix referenced above and post your results here?

CaptainMetal, 2016-12-03:
thank you. it seems to me, that this problem is fixed now. 
still some problems with albumart on some albums, but it looks like that the download doesn't find the albumart online. may be this will be solved in the future, when instead the folder.jpg file from the music directory itself is used. 

this would be kind and also faster than downloading the covers from internet, which are also sometimes different to the "real" one. 

kind regards,
captainmetal

closed 2016-12-03 by CaptainMetal