volumio/Volumio2 #734: Set default Startup volume

exetico, 2016-10-01:
Hi' 

I can see that it was been up before - but not with the best answer.

How do add a default volume at startup? 100% is just a bit to high, for my solution. I using a USB DAC, and the volume-buttons works fine in the Volumio GUI.

Are you guys thinking adding this feature - and would you pointing out a solution for my setup.

Best Regards,

01findus, 2016-10-09:
I can confirm this!  In Version 0.979 the Volume is on 100% after Startup or reboot of Volumio with an USB DAC (Nubert A200 Speakers) connected.
The Setting of the Default Startup Volume does not seem to affect the startup behaviour.

Best Regards
Alexander

volumio, 2016-10-11:
Fixed in the latest dev version
http://updates.volumio.org/pi/volumio/1.005/volumio-1.005-2016-10-07-pi.img.zip

closed 2017-06-02 by volumio