volumio/Volumio2 #736: 0.991: Can't use airplay

heis2201, 2016-10-03:
Hi,
I am glad 0.991 release is stable and I don't have any OOM or (W)LAN issues. Great works. Thanks to all devs - looks like we're getting close!

The issue I am experiencing (actually two but I will open separate issues here), is that I am not able to stream anything from my iphone to Volumio 2. From my Iphone I do see Volumio2 as potential Receiver of the audio stream but it doesn't output anything.

volumio, 2016-10-03:
What is your output device? I2S DAC? If yes, which model?
USB? Analog out? 

Let me know

heis2201, 2016-10-03:
Rpi 1 B+
Analog out via I2S Dac: Durio Sound using the HifiBerry driver.

Playing local audio files from USB or webradio stream works fine.

volumio, 2016-10-03:
Please try to reboot. Let me know

heis2201, 2016-10-03:
I will, later today.

heis2201, 2016-10-04:
Rebooted - but that didn't help.

heis2201, 2016-10-05:
I re-flashed with current Volumio2 1.001 and airplay works just fine. 

You can close this issue, NOTABUG. Thanks

heis2201, 2016-10-12:
Maybe it is an issue:
Running 1.005, airplay from my iphone is not working.

volumio, 2016-10-12:
What device are you using? I2S DAC? Which one? USB DAC? Which one?

heis2201, 2016-10-13:
Volumio 1.007 on RPI1B+ with furios Sound I2S Dac, using hifiberry driver. Playback of local files works fine.

closed 2017-06-02 by volumio