volumio/Volumio2 #737: 0.991: UPnP renderer/ DLNA server not functioning

heis2201, 2016-10-03:
On previous versions of Volumio (1.55 and betas and RC's of Volumio2), I used `pulseaudio-dlna` from [here](https://copr.fedorainfracloud.org/coprs/cygn/pulseaudio-dlna/) to stream audio from my GNU Linux to the Volumio Player. That used to work fine. Now, it detects Volumio but upon selection of that output device, Volumio remains silent.

Is that a Volumio2 or a pulseaudio-dlna issue? How to debug?

volumio, 2016-10-03:
Do:
sudo journalctl -f 

then send the stream.

Paste the log and attach them here. thanks

heis2201, 2016-10-03:
Thanks! I will do so, later today.

heis2201, 2016-10-04:
Logs attached

heis2201, 2016-10-05:
Here is an about with good news. I flashed 1.001 and tried the same thing again (using pulseaudio-dlna to stream from a GNU Fedora to Volumio2), and it works perfectly fine.

Log here:
$ sudo journalctl -f
-- Logs begin at Wed 2016-10-05 02:13:48 UTC. --
Oct 05 02:26:56 volumio volumio[803]: info: [1475634416656] InterfaceWebUI::pushState
Oct 05 02:26:56 volumio volumio[803]: info: [1475634416724] CoreStateMachine::stPlaybackTimer
Oct 05 02:26:56 volumio volumio[803]: info: [1475634416748] ------------------------------ 302ms
Oct 05 02:26:56 volumio volumio[803]: info: Pushing Favourites {"service":"mpd","uri":"","favourite":false}
Oct 05 02:27:02 volumio volumio[803]: info: CALLMETHOD: audio_interface upnp stopUpnpPlayback [object Object]
Oct 05 02:27:02 volumio volumio[803]: info: CoreCommandRouter::executeOnPlugin: upnp , stopUpnpPlayback
Oct 05 02:27:02 volumio volumio[803]: info: Stopping playback through UPNP
Oct 05 02:27:02 volumio volumio[803]: info: [1475634422757] CoreStateMachine::setConsumeUpdateService undefined
Oct 05 02:27:31 volumio sudo[1125]: volumio : TTY=pts/0 ; PWD=/home/volumio ; USER=root ; COMMAND=/bin/journalctl -f
Oct 05 02:27:31 volumio sudo[1125]: pam_unix(sudo:session): session opened for user root by volumio(uid=0)
Oct 05 02:27:40 volumio volumio[803]: info: Mpd Status Update: player
Oct 05 02:27:40 volumio volumio[803]: info:
Oct 05 02:27:40 volumio volumio[803]: [1475634460660] ---------------------------- MPD announces state update
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460685] ControllerMpd::getState
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460688] ControllerMpd::sendMpdCommand status
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460691] sending command...
Oct 05 02:27:40 volumio volumio[803]: info: Mpd Status Update: player
Oct 05 02:27:40 volumio volumio[803]: info:
Oct 05 02:27:40 volumio volumio[803]: [1475634460736] ---------------------------- MPD announces state update
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460740] ControllerMpd::getState
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460763] ControllerMpd::sendMpdCommand status
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460767] sending command...
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460806] parsing response...
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460816] ControllerMpd::parseState
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460830] ControllerMpd::sendMpdCommand playlistinfo
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460849] sending command...
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460875] parsing response...
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460885] parsing response...
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460889] ControllerMpd::parseState
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460912] ControllerMpd::sendMpdCommand playlistinfo
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460920] ControllerMpd::parseTrackInfo
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460940] sending command...
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460951] ControllerMpd::pushState
Oct 05 02:27:40 volumio volumio[803]: info: CoreCommandRouter::servicePushState
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460977] CoreStateMachine::syncState
Oct 05 02:27:40 volumio volumio[803]: info: [1475634460980] CorePlayQueue::getTrack 0
Oct 05 02:27:41 volumio volumio[803]: info: STATE SERVICE {"status":"play","position":0,"seek":487,"duration":261,"samplerate":"44.1 KHz","bitdepth":"24 bit","channels":2,"random":false,"updatedb":false,"repeat":false,"isStreaming":false,"title":"Zealots","artist":"Fugees","album":"The Score","uri":"USB/SONY_8GM/Hip Hop/Fugees-The_Score [1996]/04^ Fugees - Zealots.mp3","trackType":"mp3"}
Oct 05 02:27:41 volumio volumio[803]: info: CURRENT POSITION 0
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461015] CoreStateMachine::syncState   stateService play
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461018] CoreStateMachine::syncState   currentStatus stop
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461021] CoreStateMachine::pushState
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461049] CoreStateMachine::getState
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461067] CorePlayQueue::getTrack 0
Oct 05 02:27:41 volumio volumio[803]: info: CoreCommandRouter::volumioPushState
Oct 05 02:27:41 volumio volumio[803]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461089] InterfaceWebUI::pushState
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461276] ------------------------------ 628ms
Oct 05 02:27:41 volumio volumio[803]: info: Pushing Favourites {"service":"mpd","uri":"","favourite":false}
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461320] parsing response...
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461339] ControllerMpd::parseTrackInfo
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461349] ControllerMpd::pushState
Oct 05 02:27:41 volumio volumio[803]: info: CoreCommandRouter::servicePushState
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461373] CoreStateMachine::syncState
Oct 05 02:27:41 volumio volumio[803]: info: [1475634461388] CorePlayQueue::getTrack 0
[... cut ...]
Oct 05 02:28:05 volumio volumio[803]: info: CALLMETHOD: audio_interface upnp stopUpnpPlayback [object Object]
Oct 05 02:28:05 volumio volumio[803]: info: CoreCommandRouter::executeOnPlugin: upnp , stopUpnpPlayback
Oct 05 02:28:05 volumio volumio[803]: info: Stopping playback through UPNP
Oct 05 02:28:05 volumio volumio[803]: info: [1475634485091] CoreStateMachine::setConsumeUpdateService undefined
Oct 05 02:29:12 volumio systemd[1]: Starting Cleanup of Temporary Directories...
Oct 05 02:29:12 volumio systemd[1]: Started Cleanup of Temporary Directories.

I guess, we can close this issue. **NOTABUG**

closed 2016-10-12 by heis2201