volumio/Volumio2 #738: 1.001 hotspot not reachable with r8188eu

balbuze, 2016-10-04:
Hotspot can't be see even if it appears actived in Volumio (and other networks can be seen ). This usb dongle is able to manage hotspot, I saw it working in a far previous version.
driver r8188eu

macmpi, 2016-10-05:
Similar issue here with TL-WN725N & built-in legacy r8188eu driver. 
Many disassoc message on console.
Wifi won't connect on my AP (Network UI does not see SSIDs). Sometimes pretends Hotspot is created, but actually it does not exist for real.

Haven't tried alternate driver suggested by [gogeta here](https://volumio.org/forum/volumio-final-release-very-near-pre-test-needed-t4709-50.html), but wifi used to work fine with legacy driver.

macmpi, 2016-10-05:
Noted many components may not be up-to-date (apt-get upgrade), including wpasupplicant. Have not tried upgrading yet.

volumio, 2016-10-05:
All the packages are installed when we create the image, so they are reasonably up to date. If someone is not, its because its a custom compiled version, which we need.
Apt-get upgrade is not recommended, especially if you want to use the OTA update

volumio, 2016-10-06:
Added here https://github.com/volumio/Build/commit/251d5bc9f9edcac4e210b85f8b658ac4da824616

macmpi, 2016-10-06:
Thanks.
Actually installed manually but still no success.
I then upgraded to 1.003, still no luck
Manual Wifi connection panel  refresh does nothing, and no SSID listed

Not sure it is a driver issue.
Baseline driver from kernel was working so far in RC2 and following non-test releases updates. First test release I tried was 1.001, so can't tell exactly when it broke in test releases.

macmpi, 2016-10-06:
Ok, made a bunch of tests pulling-up another SD-card and starting fresh.
I confirm Volumio with TL-WN725N connects fine to wireless APs using stock r8188eu driver up to version .979.
Upgrading by OTA from there towards 1.003 (or 1.001) kills wifi operation.
Making a fresh SD directly to 0.998 has same problem.
It seems there is some conflict between hotspot functionality and normal wifi operation: unsure at which point this function was introduced, or significant changes made.

In most cases, turning OFF hotspot function restores wifi operation with std driver.

MrEngman's driver 8188eu allows use of wifi with hotspot function ON, though I did not fully test the hotspot function yet.
However I noticed that, wifi seemed, in my case, less reliable with that custom driver, and UI seems to report misleading signal strength when used.

Would be interested by other users experience.

I'm not sure Hostspot can be done with std driver: need to do more search.
@volumio, is Hotspot built in a very close fashion to MrEngman's [write-up](https://www.raspberrypi.org/forums/viewtopic.php?p=948118#p948118) related to his driver? 

Also, I noted that the (new) boot sequence has a blocking fashion to raise network interface early-on (after nf_countrack version 0.5.0 message).
"A start job is running for LSB: Raise network interfaces (.../no limit)."
The no-limit happens to totally block  boot sequence if my pi only has wifi dongle...
If I plug an ethernet dongle in addition, it passes (thanks to ethernet), and then later in the sequence wifi gets up: so I have 2 interfaces up in the end.
Would rather not block at that stage, and let wifi turn on later.
Removing the ethernet dongle after boot (and letting some time for Volumio to decommission it), allows to run on wifi only as wanted...

macmpi, 2016-10-06:
hum...seems a [hostapd issue](https://jenssegers.com/43/realtek-rtl8188-based-access-point-on-raspberry-pi) rather than driver prior to 4.4.13.
[Some claim](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=152603&p=1010357#p1010357) with 4.4.13 hostapd does not need to be recompiled anymore...

macmpi, 2016-10-09:
Based on previous references, it seems rtl8188eu support in hostapd depends on nl80211 driver.
Volumio 2 built-in hostapd is version v2.3
MrEngman's hostapd is a Realtek modified version 2.4

I tried replacing hostapd by MrEngman's hostapd and added relevant `driver=nl80211` statement in hostapd.conf, but had no luck in getting hostpot working yet.
Actually it seems Volumio is probably generating hostapd.conf itself, so my changes may not be relevant. I see there is a specific hostapd-edimax.conf, maybe some magic is needed to also make a r8188eu template and handle it properly...

As it seems r8188eu support in unmodified hostapd depends on some improved nl80211 support, I looked [here](https://w1.fi/cgit/hostap/plain/hostapd/ChangeLog): there are several changes related to nl80211 in most recent 2.5 and 2.6.
Therefore, based on earlier references,  it's likely that using most recent hostapd builds (vs actual 2.3), will improve rtl8188eu support, along with other niceties...
Still there might need some tuning in hostapd.conf to get things right.

balbuze, 2016-10-09:
On Sparky 
now I can get it working : boot with ethernet plugged, I changed channel to 1,disable password, save and I can see hotspot ! Unplug the ethernet cable and it still works. But after reboot everthing is lost.
So driver is ok...
I'm going to check with rpi b and rpi 2 (not the same driver)

balbuze, 2016-10-09:
ok with rpi b : work now as I updated the driver according to https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=62371&sid=320e49a482156ed7f1ebc88f078af5a8 the hotpsot is created and reachable. In webUI I see only the hotspot with ip 192.168.211.1 and whatever the ip adress in my webrowser I join the device.
I have to test with RPI 2 now

macmpi, 2016-10-09:
On my rpi0, the issue with that MrEngman's driver is that although it may provide hotspot feature, rpi can not join AP anymore, when hotspot function is OFF.
Network panel won't list any AP, & typing SSID info on connect UI, leads to nowhere. Wireless restarts, and network panel reports hotspot IP again (though hotspot section is OFF).

Only way to have normal wifi operation (i.e. rpi join AP), is to use the std driver.
I thing we'd be better using the std driver and sort-out the hostapd issue (special version, or newer). Still, would need to know how Volumio invokes hostapd to pass-on proper config file with appropriate driver name.

macmpi, 2016-10-11:
Ok, understood the MrEngman's driver issue in my setup, and confirms it works for normal wifi access, and seems only option for Hotspot function.
There's been several conflicting issues recently in network handling like [issue #753](https://github.com/volumio/Volumio2/issues/753) which made this complex to debug, as I updated Volumio in the meanwhile...

So it's likely including MrEngman's driver in next builds is necessary.
Then to get hostapd working with that driver, we might need to have `driver=nl80211` statement in relevant hostapd.conf file.
It's unclear at this point if under 4.4.9 (used by Volumio2), generic hostapd can properly handle encryption: custom-build hostapd may be necessary. It seems that on later kernel versions, generic is just fine.

Let's review that post 1.005 once issue #753 is resolved and other .conf considerations are handled.
Sorry for confusion, but hope this helps in the end.

volumio, 2016-10-11:
I'm trying the same dongle with Mr Engman's drivers... But I can't get them to connect ... This will be a tricky one..

macmpi, 2016-10-12:
Had some issues too. I decided to validate the overall setup from a blank Jessy lite image (as there are still some network setting issues with 1.005).
I had difficulties to get MrEngman's driver work even for base wifi operations, and finally discovered some issues due to some hiccups with USB hub in-between!...
Very strange: if you use a USB hub (that is known good & even working with base driver), try changing ports with dongle. As weird as this may seem, it [ended up working](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=62371&start=1125): USB handling in rpi still has some voodoo involved...

With that done, I moved to hostapd, and followed [MrEngman's tutorial here](https://www.raspberrypi.org/forums/viewtopic.php?p=948118#p948118), EXCEPT that I did NOT install his custom hostapd build which is pre 4.4 based.
It turns-out on 4.4.23+ (my Jessy image), baseline hostapd works great, with WPA authentication: I type this message on my Mac, using my rpi0 as wifi AccessPoint :)

I'm eager to replicate this onto Volumio now, but I'd need a post 1.005 release that fixes issue #753 and allow to set hostapd.conf with `driver=nl80211`.
Don't know for sure if 4.4.9 will be sufficient for WPA support with baseline hostapd: I read info about 4.4.13, but worth a try.

volumio, 2016-10-12:
What exactly you need fixed? To be able to disable the hotspot? And add the driver=nl80211 to hostapd.conf? 
To do that we need to make another special exception for the chipset that this dongle uses ... 
Could you send me the driver you compiled please?

macmpi, 2016-10-12:
Yes it would be useful to be able to turn hotspot ON/OFF in order to fully test both modes  (unsure what happened between 1.003 and 1.005 on this), and have hostapd-8188eu.conf as follows:
```
interface=wlan0
driver=nl80211
bridge=br0
ssid=volumio
channel=1
wmm_enabled=0
wpa=1
wpa_passphrase=Volumio2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
auth_algs=1
macaddr_acl=0 
```
I did not compile any driver, only used MrEngman's which happen to hav necessary nl80211 support built-in. So the one you initially included in build scripts should be fine.

Then I'm not sure if you explicitly bridged the interfaces with bridge-utils, but then **/etc/network/interfaces** should look like:
```
auto lo
iface lo inet loopback

iface eth0 inet dhcp

auto br0
iface br0 inet dhcp
bridge_ports eth0 wlan0
```

Does it help?
I'm still trying to fiddle sth out of 1.005, but have some issues with network config setup.

volumio, 2016-10-12:
Let's leave out the bridge part for now... 
I 'll try to add the edits you asked

macmpi, 2016-10-12:
Isn't the bridge stuff easier to setup than NAT & stuff ?
Do we really need more than a bridge for the Hotspot feature?

macmpi, 2016-10-13:
We are getting closer with 1.008, and bundled MrEngman's driver: thanks a lot for your efforts!

Normal wifi seems to work great when hotspot is OFF.
I may have a non std config though (due to lots of trial/errors in previous versions), as my **/etc/network/interfaces** is the std Jessy one
```
# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

iface eth0 inet manual

allow-hotplug wlan0
iface wlan0 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

allow-hotplug wlan1
iface wlan1 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

```
Volumio may use different settings. I may need to do a factory reset and make more tests.

Hotspot function does not seem fully working yet (SSID visible, but login failed): it seems you did not provide a hostapd.conf made for 8188eu as mentioned earlier?

I simply added `driver=nl80211` in /etc/hostapd/hostapd.conf and restarted hostapd by `sudo hostapd -dd /etc/hostapd/hostapd.conf` and... voilà!
So you just need to add that line to a dedicated hostapd.conf for that wifi module, and we should be very close!

Current implementation of Hospot seems very much to turn Volumio into a router (with NAT, DHCP,...): I imagine this is making code, settings, memory under lots of stress.
Wouldn't AP as a bridge be sufficient, as zeroconf addressing should be enough for Airplay?
And if linked to one port to a network with DHCP, it would benefit from existing subnet address management.

macmpi, 2016-12-05:
@volumio 
> I'm trying the same dongle with Mr Engman's drivers... But I can't get them to connect ... This will be a tricky one..

Did you have Edimax dongle at same time? Maybe both are conflicting due to [possible driver conflicts](https://github.com/lwfinger/rtl8188eu/issues/185). However we should expect average user might only have only one at a time.

MrEngman's 8188eu driver seems now fine with TL-WN725N and clones: you may add back previously [reverted commit](https://github.com/volumio/Build/commit/251d5bc9f9edcac4e210b85f8b658ac4da824616) (maybe using kernel version in a variable, to avoid future issues if Volumio Kernel gets updated at some point).
To get hotspot working with that driver (and with actually most wifi adapters, except Edimax which  is indeed treated as a special case), please add `driver=nl80211` in baseline [hostapd.conf](https://github.com/volumio/Build/blob/master/scripts/volumioconfig.sh#L408), which seems the most standard configuration for hostapd.

With that we should hopefully be able to close that one now.

macmpi, 2017-03-01:
MrEngman drivers [added](https://github.com/volumio/Build/commit/d9c908b2f4743e8116bc82494c8f7a64cb95d63f): should close this longstanding issue, together with previous one related to hostpad.conf.

@balbuze you may confirm & close with next image release.

closed 2017-06-02 by volumio