volumio/Volumio2 #741: Set Time and Date correctly using NTP service (timedatectl)

heis2201, 2016-10-05:
Looks like Volumio2 does not enable NTP by default.
Noticed that time was not set correctly, so had to run 
`timedatectl set-timezone EST` and then
`timedatectl set-ntp true`

Maybe we should include the first step as an input field in the GUI and the enable `ntp.service` by default

macmpi, 2016-10-20:
Would definitely be nice to have some UI in System pref panel to properly set date/time/timezone indeed, as great new timed features (Sleep and Alarm) are difficult to set without known time reference.
Thanks!

xipmix, 2016-10-25:
See also #373

heis2201, 2016-10-25:
Not a programmer here, so likely that I don't get the challenge described in #373 but

`timedatectl list-timezones > export-tz.txt` gives you a complete list of available timezones that can be used with `timedatectl set-timezone ...`

Why not have a select box with this entire list and upon *Save timezone* the selected string is send to little bash script that does `sudo timedatectl set-timezone selected_string`.

biva, 2017-03-15:
Hello, I tried the above operations, but I get an error: "Failed to set time zone: Access denied"
I used the password "Volumio".
So I can't change the timezone: what am I doing wrong?
Thanks!

macmpi, 2017-03-15:
how about `sudo dpkg-reconfigure tzdata`?

biva, 2017-03-15:
Great, that works! Thank you @macmpi !

macmpi, 2017-03-24:
but https://github.com/volumio/Volumio2/issues/741#issuecomment-256066520 is more systemd-friendly for setting this up in a future UI widget