volumio/Volumio2 #742: Use of "internal storage" for .pls /.m3u internet radio stream files

heis2201, 2016-10-05:
Copied all my custom .m3u and .pls internet radio stream files from previous Volumio 1.55 installation to volumio2ip/internal%20storage/ . They are listed in the UI, I can browse them under Library\INTERNAL but they don't play, none of all.

buzink, 2016-12-26:
As a workaround, you can create playlist files in the folder "/data/playlist" following the following format:

[ { "service": "webradio", "uri": "http://icecast.omroep.nl/radio1-bb-mp3", "title": "Radio 1", "albumart": "/albumart?path=%2Fplaylist%2Fradio1.jpg" } ]

You can add playlists through SSH each time or through your file explorer after adding "/data/playlist" once as a network share in "/etc/samba/smb.conf"

You can also convert podcast rss-feeds to Volumio 2 playlist files, see https://github.com/buzink/podcast2playlist.

thelieisajoke, 2017-05-10:
THANK YOU buzink!
this should be stickied in the forums

heis2201, 2017-06-27:
that's a lot of manual work when you want to import a bunch of radio stations, especially since their .pls/.m3u files have all the information needed.