volumio/Volumio2 #746: 1.003 blocked startup sequence: due to /etc/network/interfaces ?

macmpi, 2016-10-07:
Hi,

I referred to this issue [in here,](https://github.com/volumio/Volumio2/issues/738) but thought it should be separated-out.
In startup sequence, system tries to endlessly raise network interfaces, and it seems it may block the whole startup if the pi only has 1 wifi interface.
It seems similar to what is [discussed here](https://ubuntuforums.org/showthread.php?t=2323253)
2 approaches for fix: set a timeout, or let interface be hotplug
Timeout will still delay the whole startup sequence so I looked at the latter.

I applied the suggested change to /etc/network/interfaces, i.e. commenting-out "auto" statements:
```
#auto wlan0
auto lo
iface lo inet loopback

#auto eth0
allow-hotplug eth0
iface eth0 inet dhcp

allow-hotplug wlan0
iface wlan0 inet manual
```
Now all is fine, and pi is doing its whole startup sequence fine, and I get wifi in the end.
Unsure if this may conflict with other stuff linked to hotspot setup which I've not played with so far, but definitely worth a look.

volumio, 2016-10-07:
Yes, auto shoould not be needed on ethernet. We added it because we found that without this, the eth0 interface was not brought up automatically (must be something with udev). So, in order to remove auto eht0 we must find a way to fix the above before. Could you help?

xipmix, 2016-10-07:
auto eth0 will block until the network is up, which in most cases is what you want; allow-hotplug allows things to start in parallel. Maybe that is affecting the ordering of events that occur during and after network bringup?
Something that may help thinking about this - how does systemd know that the things that need the network can be started? And that _all_ the interfaces are up? Historically it has been a problem so it has its own stuff to handle dhcp etc. This might be more reliable in the long run. This blog post may be helpful https://www.joachim-breitner.de/blog/664-Switching_to_systemd-networkd.

macmpi, 2016-10-07:
I'm a bit confused here.
For reference, I'm using a rpi0, therefore all network interfaces come from USB adapters...unsure it makes a difference.
With the 2 auto statement commented-out as in first post, the 2 interfaces are up at the end of the startup sequence if both are connected on the device.
Therefore I get a no-brainer experience, with network interfaces up if and when they are available and plugged-in. Nothing blocked if they are not available.
Unsure why it would be different on other raspi having on-board interfaces.

macmpi, 2016-10-14:
Hi,
After 2.000 upgrade (and restore factory settings), same issue, as there is still `auto wlan0` statement in initial **/etc/network/interfaces**, which leads to unavailable interfaces (eth0 remains in zeroconf, wlan0 not inited).

Removing `auto wlan0` fixes the problem.

You may want to [fix it there](https://github.com/volumio/Volumio2/blob/f67e9a9a208a3f4922d29d3433affed1c128408a/app/plugins/system_controller/network/index.js#L556), as you did for eth0.

volumio, 2016-10-14:
Sorry, but I don't get it, what is the problem resulting from this?

macmpi, 2016-10-14:
With original **/etc/network/interfaces** starting by `auto wlan0`, rpi0 is not reachable on the home network after reboot: ethernet dongle IP is on zeroconf (169.254.x.x), and wlan0 is not available.
Removing that only statement, rpi0 networking is ok after reboot (ethernet & wlan get the dhcp address on home network).

It seems `auto eth0` has been removed in  Volumio2/app/plugins/system_controller/network/index.js after original report in this issue, but not `auto wlan0`.
It looks like both need to be removed in my setup.

Does it make sense?

volumio, 2016-10-14:
Yes it does, let me check how it behaves and then I'll do it!

gpgekko, 2016-10-15:
Is this also what is discussed at https://volumio.org/forum/can-access-webui-t4748-20.html#p22118?

macmpi, 2016-10-15:
yes most probably

macmpi, 2016-12-08:
fixed as [2.036](https://volumio.org/forum/unable-install-volumio-disk-image-zero-t4936-40.html#p25097). Thanks!

volumio, 2016-12-08:
I introduced some more fixes here:
http://updates.volumio.org/pi/volumio/2.039/volumio-2.039-2016-12-08-pi.img.zip

Could you let me know if that works as supposed?

closed 2016-12-08 by volumio

macmpi, 2016-12-08:
Yes saw changes on dhcpcd and updated 2.036 to 2.039 right-away ;)
Still working with original 2.036 setup, but will burn a new 2.039 and let it first-boot to see.

macmpi, 2016-12-08:
Fresh 2.039: all good on rpi0 with only USB-wifi dongle (no eth).
Still some dhcpcd errors in journalctl, but can connect to network resources and play web radios.
```
Dec 08 15:17:23 volumio dhcpcd[473]: Not running dhcpcd because /etc/network/interfaces ... failed!
Dec 08 15:17:23 volumio dhcpcd[473]: defines some interfaces that will use a DHCP client ... failed!
```