volumio/Volumio2 #753: 1.005 turning off hotspot in webui does not turn it off

xipmix, 2016-10-10:
Preparation: Fresh volumio 1.003, updated by turning on test mode in /dev and then checking for updates. Reboot.

I want to be able to turn off the hotspot. When I click on Menu->Network->EnableHotspot=Off,
 the button updates ok and in /data/configuration/system_controller/network/config.json I get
```
  "enable_hotspot": {                                                           
    "type": "boolean",                                                          
    "value": false                                                              
  },                                                                            
```
But at the top of the display, the wireless hotspot still shows up as online with the hardcoded 192.168.211.1 address. ifconfig shows the same ip address and the interface is still 'up'.

Reload browser, go back to Menu->Network, and the hotspot is still shown as active in the status area at the top of the page, but as turned off  in the setting at the bottom.

Next I set wirelessnetwork: automaticip=no, and the static config items to undefined. Reboot. It still starts the wlan interface and the status area shows the hotspot as active.

At this point:
```
$ cat /etc/network/interfaces
auto wlan0
auto lo
iface lo inet loopback

allow-hotplug eth0
iface eth0 inet dhcp

allow-hotplug wlan0
iface wlan0 inet manual
```
When I reboot with the system in this state, it takes about 40s before the web UI appears. It is waiting for the wireless hotspot to be started up; before the UI is there, wlan0 is not up.

As far as I can tell what is going on is:
 * /lib/systemd/system/wireless.service is calling 
 /volumio/app/plugins/system_controller/network/wireless.js with the argument 'start',
 * This is supposed to run 'stopHotspot', 'stopAP', and then 'startFlow'.
But 'startFlow' calls 'startAP' without first checking the value of enable_hotspot
and as far as I can see startAP does not check either.
startAP runs ifdeconfig (which should take down the wlan interface if it is up) and then wpasupplicant, followed by dhclient.
  * /etc/wpa_supplicant/wpa_supplicant.conf is empty (has a single space in it)
  * /data/confguration/wlanstatic does not exist.

* I guess that later startFlow gets to where it calls startHotspot, which starts by calling stopHotspot and then starthostapd, again without checking the enable_hotspot setting.

This would be easier to understand if I could get the console.log() output. How do I turn that on?

volumio, 2016-10-11:
You got the whole thing! To see logs:
sudo journalctl -f 

(you can grep it to see the relevant part... for example
sudo journalctl -f | grep wireless )

How do you suggest to change all the above?

xipmix, 2016-10-12:
Silly me, I tried dmesg and volumio.log but forgot journalctl, thank you.

Where to start... What I think makes sense to do here is check enable_hotspot, in the startFlow() function. If it's false then just log that and return. something like:
```
+ if (! enable_hotspot) {
+     console.log("Hotspot disabled in volumio configuration.");
+     return;
+ }
console.log("Start wireless flow");
```
But it's probably more complicated than that...

It looks from the log like wireless.js is starting quite early, before the main volumio process which suggests to me it is starting even before  /data/configuration/system_controller/network/config.json has been read. If so, how do we get the configuration information?

Question: If the hotspot is enabled, is that wireless interface intended to be just listening for clients, or is it intended that the same interface can be a wireless network client as well (different ssid, different subnet)? I don't know if this is even possible but if it's supposed to work that way then the logic for disabling the hotspot may be more complicated still.

SIde note: it looks like the single space in /etc/hostapd/hostapd-edimax.conf (and presumably /etc/hostapd/hostapd.conf too)  are errors - if I get time I'll look at the volumio build scripts.
```
Oct 10 13:33:56 volumio hotspot[531]: 1 errors found in configuration file '/etc/hostapd/hostapd-edimax.conf'
$ od -x  /etc/hostapd/hostapd-edimax.conf
0000000 0020
0000001
```

xipmix, 2016-10-12:
Attached output of 'sudo journalctl -u wireless', shortly after boot.
[wireless.zip](https://github.com/volumio/Volumio2/files/524143/wireless.zip)

volumio, 2016-10-12:
Well, we could directly read the config.json file... I don't like it since it will be a little bit "detached" but it should work.

macmpi, 2016-10-12:
1.007 and same issue.
Hotspot appears off on UI, but wlan interface IP is still 192.168.211.1

It seems the whole hotspot feature runs full NAT stack & stuff, which makes config much more tricky.
Is that mandatory?
Couldn't it be a simple bridge, with no NAT, no DHCP service?

xipmix, 2016-10-13:
I understand your concern about the 'detached'ness. What if you wait until volumio itself starts and has read the configuration as it normally would, then check if the hotspot needs to be turned off again? Also ugly but perhaps less so.

Taking that line of thought a bit further... The reason I asked about dual use of the wlan interface above is that - if the hotspot service is part of volumio,  why not wait until volumio is 'up' before trying to activate it (e.g. near the time the volumio sound is played), rather than asking systemd to do it during bootup? You could tell systemd what dependencies the hotspot has so that when volumio tries to start the hotspot, systemd brings those other services up first.

volumio, 2016-10-13:
The hotspot feature is embedded in volumio's wireless networking, and that must start before volumio, if not there would not be connection when volumio starts up.

I am about to start implementing the complete turn off of the hotspot. In fact, what we have (when we disable it) is just that wireless interfaces is set to a particular ip, but the hotspot itself is not started. I agree that this leads to confusion.

Could you please send me the link of the wifi driver that you used (mr engman's one), I am not able to connect with the one I found...

macmpi, 2016-10-13:
Here (I guess you refer to th eother issue #738 :
https://dl.dropboxusercontent.com/u/80256631/8188eu-4.4.9-884.tar.gz or
https://dl.dropboxusercontent.com/u/80256631/8188eu-4.4.9-v7-884.tar.gz for raspi2/3

volumio, 2016-10-13:
Ok, I made it see https://github.com/volumio/Build/commit/a2800e65b83b34558349ab6e9236905538a01672

Building a new image now

macmpi, 2016-10-13:
ok, thanks!
1.008 seems to work better for hostapd switch.
Hotspot not fully functional with my dongle yet, but I follow-up on related issue #738

xipmix, 2016-10-15:
This is fixed in 2.000. Many thanks! How I tested is below.

As before I  turned off hotspot in the webui. Nothing happened (at least in the logs, journalctl -f) until I pressed the 'save' button. The status indicator still showed hotspot active at this point. When I reloaded the browser and went back to the network page the hotspot status display (at the top of the page) was not showing any more. eth0 also loses its address and has to get it back again from dhcp. At this point I ran 'ifconfig' and it showed wlan0 was down. Rebooted and ifconfig still showed wlan0 as down. On the network page the hotspot status display is not visible. Thanks once again.

volumio, 2016-10-15:
Thanks to you who helped fixing that! !

closed 2016-10-15 by volumio