volumio/Volumio2 #759: Volumio Web App does'nt work

akhlan, 2016-10-12:
Hi,

I use Volumio 2 with a Rpi 3 + digidac and impossible to connect on it with my Android phone and "Volumio Web App" and you ?

If it's impossible for you to, how control Volumio with my phone ?

Rgds

pilifsch, 2016-10-12:
Please provide more details. Did you managed to connect to Volumio at least once? How about PC? Are you connecting via Volumio's hotspot or is it connected to your router by cable or WiFi? What address did you use to connect? Did you change Volumio's name?

akhlan, 2016-10-13:
I use a Rpi 3 by cable, my music is stocked on my NAS Synology, I use http://volumio.local with a MAC or MPDroid with my LG G4 and all is fine... I would like use "Volumio Web App" with my LG G4 because MPDroid makes duplicate empty files but I don't know why, it's impossible for me...

pilifsch, 2016-10-13:
Did you try using Volume's IP address?

volumio, 2016-10-13:
http://volumio.local does not work on Android. You must use the IP address...
If you're referring to an android app, this was made for old volumio and will not work anymore...

pilifsch, 2016-10-13:
Exactly, just found it here: https://volumio.github.io/docs/User_Manual/Quick_Start_Guide.html

jongfeli, 2016-10-13:
> If you're referring to an android app, this was made for old volumio and will not work anymore...

I think @akhlan is talking about the Android app, called [**Volumio Web App**](https://play.google.com/store/apps/details?id=com.volumio.moritz.volumiowebapp_release). I also tried the app but now I understand why it did not work. 

In the Play Store there is no mention of any kind of the Volumio version needed for it to work. Connection via a web browser works just fine.  

Are there any plans to getting the **Volumio Web App** working also with **Volumio 2**?

Regards

akhlan, 2016-10-13:
Excatly, I spoke about the Android App ;) I'll wait the new version so ;) MPDoid (Android App) don't work fine for me...

PS : I use Volumio2 IP address on my phone...

volumio, 2017-06-13:
Volumio webapp does not work with volumio2, use the official app instead

closed 2017-06-13 by volumio

akhlan, 2017-06-13:
I don't find official app on the Volumio website ><

volumio, 2017-06-13:
There you go:
https://play.google.com/store/apps/details?id=volumio.browser.Volumio&hl=it