volumio/Volumio2 #760: WIFI wont reconnect after AP restart

fspralja, 2016-10-13:
After AP/Router restarts volumio2@pi3 goes into "Volumio" AP mode and wont reconnect to the original AP when it gets available. To fix it only solution is to hard restart the pi which is bad or connect to the Volumio wifi and restart from the UI.

Any chance the auto AP mode can be disabled somehow?

pilifsch, 2016-10-13:
I have observed the same issue.

xipmix, 2016-10-13:
This sounds like #753 but it would help to understand the issue if you could post a log file.
```
journalctl  >volumio.log
zip volumio.zip volumio.log
```
and attach the .zip here.

pilifsch, 2016-10-13:
Here you go. Just to be clear:
1) Restarted Volumio and waited for successful boot and connection to my home router
2) Restarted router
3) Volumio started it's own hotspot, so I connected to it using my phone
(I don't know why, but I can't access Web UI when I connect via PC - it can't get get IP address my iPhone shows Webpage automatically after connecting to hotspot - as if it was login window to access Internet connection)
4) Then I manually reconnected to my router (to get log file via ssh using my computer)

[volumio.log.zip](https://github.com/volumio/Volumio2/files/526943/volumio.log.zip)

volumio, 2018-03-26:
Done

closed 2018-03-26 by volumio