volumio/Volumio2 #766: Play an entry from favourites

patrickse, 2016-10-14:
I am currently writing an addon for openhab2 to support volumio2. I already implemented the basic features like volume and playback control + showing current artwork + title, album and artist. 

Now I want to implement a push button solution to start playing an entry from my favourites. For example a radio station.

I found the command playRadioFavourites which actually adds all my Radio Favourites to my queue. What I would like to achieve is, pass a **playRadioFavourite** command with an index or name of the entry and volumio could start playing this entry.

Is there a way to implement it at the moment?

volumio, 2016-10-14:
Just pass:

'play', N

where N  is the number in the queue that you want to play.

Or, you can pass directly the URI of your favourite radio...

patrickse, 2016-10-14:
Oh ... that easy ;) that´s working quite well. The only problem I´ve now facing is... call playRadioFavourites populates the favourite entries to the queue.. But the new entries in the queue all have the label undefined. 

I am on the newest 2.0 release

volumio, 2016-10-14:
This API call is unused, so it might not work properly. I suggest to add just the URIs that you're interested to...

patrickse, 2016-10-14:
ok.. but I actually like the way that volumio manages the different media sources and openhab just remote controls it...

Thanks anyway... I will stick for now with this playRadioFavourites command. Hopefully you will not deprecate it :)

volumio, 2016-10-14:
No we won't, we can also improve it... Feel free to send a PR to add the functionality you need!

patrickse, 2016-10-14:
👍 thanks. I will come back to this later...

closed 2016-10-14 by patrickse

Cedric-NET, 2019-03-29:
hello,
this dates back to 2016 but I'm now callin from 2019 and trying to use playradiofavourites

It does not seem to work at all, was it removed ?

Is there a way to send api calls (rest, console or other) to play from the radio favorites ?

thanks !