volumio/Volumio2 #774: NFS mount problem

cyclopse11, 2016-10-16:
I have moved onto version 2 from 1.55. Everything is working except my NFS file share. Entered the details exactly as they were for 1.55. Is there anything I can look into to debug?

Many thanks

Steve

Satyr77, 2016-10-16:
I noticed that NFS shares are not correctly added when using an alias with a blank. Then all following shares that are correctly added are not displayed and can't be changed or deleted. The shares that are not working are displayed in the libary, too

tpar7322, 2016-10-17:
I had a similar issue. I found that changing from CIFS (which is the default option) to NFS **before** entering the other details worked for me. Not sure why. YMMV.

xipmix, 2016-10-17:
@cyclopse11 you should be able to see what is happening in the system log. Try rebooting your volumio, then connect to the web ui and try adding your share as the first thing you do. Then, connect via ssh and save the log:
```
% ssh volumio@volumio.local
$ sudo journalctl >addshare.log
$ exit
% scp volumio@volumio.local:addshare.log .
```
This will give you a log file with the events relating to the share addition near the end (look for 'addShare'). You can also watch the log in realtime
```
% ssh volumio@volumio.local
$ sudo journalctl -f
...lots of log stuff
^C
$ exit
```

cyclopse11, 2016-10-17:
Great suggestion tpar7322 but didn't work here. 
xipmix here is the output after trying to mount the NFS drive;-
Oct 17 17:08:59 volumio volumio[782]: info: Adding a new share
Oct 17 17:08:59 volumio volumio[782]: info: No correspondence found in configura                                                                             tion for share Bulldozer on IP 192.168.1.7
Oct 17 17:08:59 volumio sudo[1157]: volumio : TTY=unknown ; PWD=/ ; USER=root ;                                                                              COMMAND=/bin/mount -t nfs -o ro,soft,noauto 192.168.1.7:e/music /mnt/NAS/Bulldoz                                                                             er
Oct 17 17:08:59 volumio sudo[1157]: pam_unix(sudo:session): session opened for u                                                                             ser root by (uid=0)
Oct 17 17:08:59 volumio sudo[1157]: pam_unix(sudo:session): session closed for u                                                                             ser root
Oct 17 17:08:59 volumio volumio[782]: exec error Error: Command failed: /usr/bin                                                                             /sudo /bin/mount -t nfs -o ro,soft,noauto 192.168.1.7:e/music /mnt/NAS/Bulldozer
Oct 17 17:08:59 volumio volumio[782]: mount.nfs: mounting 192.168.1.7:e/music fa                                                                             iled, reason given by server: No such file or directory

Many thanks
Cyclopse11

xipmix, 2016-10-17:
> Oct 17 17:08:59 volumio volumio[782]: mount.nfs: mounting 192.168.1.7:e/music failed, reason given by server: No such file or directory

Looks like the path you gave 'e/music' is bad. I think you need the full path with a leading '/',
so perhaps '/e/music' will work.                                     
                                                                     
If you can login to the nas and run 'showmount -e' that should show the directory paths that are exported.

cyclopse11, 2016-10-18:
Hi xipmix,

Tried /e/music without success. It looks like e:\Music in Hanewin NFS server. (I changed it to a capital M to reflect the windows folder). So in Volumio it now looks like 192.168.1.7:e/Music but still refuses to mount. 
Please note I have no issues with Volumio 1.55 with mounting the Music folder. 

The Windows path is E then the Music folder. Think I am running out of options now.

xipmix, 2017-02-19:
Hi, this may have been fixed by #1010. When a new volumio release is made, please can you test and report back? You will need a version higher than 2.041.

closed 2017-06-02 by volumio