volumio/Volumio2 #778: v2.002, odroid: Edit network drive doesn't work

RastaX, 2016-10-17:
Latest version of Volumio on OdroidC2 with HifiShield:

TAB My Music:
The drive is mounted (mounted-status: green, size: 1.42TB) on a wired 1gb/s network smb-share (samba on raspbian).
I want to change the path from 192.168.170.12:drive2/_musiklib to 192.168.170.12:drive2/_templib.
After clicking "Edit Drive" and making the changes in the textbox, which are instantly previewed above 8), I hit save. The changed path in the "Network Drives" list stays as previewed, but reverts back after 2 seconds.

Any ideas?

RastaX, 2016-10-17:
I've found  "sudo journalctl -f" :)

Oct 17 18:19:02 volumio volumio[904]: [1476728342598] ---------------------------- Client requests Menu Items
Oct 17 18:19:02 volumio volumio[904]: info: [1476728342609] ------------------------------ 11ms
Oct 17 18:21:26 volumio volumio[904]: info: CoreCommandRouter::getUIConfigOnPlugin
Oct 17 18:21:26 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:21:26 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: networkfs , listShares
Oct 17 18:21:30 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:21:34 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:21:38 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: networkfs , editShare
Oct 17 18:21:38 volumio volumio[904]: { path: 'drive2/_musiklib',
Oct 17 18:21:38 volumio volumio[904]: ip: 'NAS',
Oct 17 18:21:38 volumio volumio[904]: id: 'SCRAMBLED_ID_FIGURES',
Oct 17 18:21:38 volumio volumio[904]: name: 'elements2',
Oct 17 18:21:38 volumio volumio[904]: fstype: 'cifs',
Oct 17 18:21:38 volumio volumio[904]: username: 'MYUSERNAME',
Oct 17 18:21:38 volumio volumio[904]: password: 'MYPASSWORD',
Oct 17 18:21:38 volumio volumio[904]: options: '',
Oct 17 18:21:38 volumio volumio[904]: mounted: true,
Oct 17 18:21:38 volumio volumio[904]: size: '1.42 TB' }
Oct 17 18:21:38 volumio sudo[1592]: pam_unix(sudo:auth): conversation failed
Oct 17 18:21:38 volumio sudo[1592]: pam_unix(sudo:auth): auth could not identify password for [volumio]
Oct 17 18:21:38 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:21:39 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: networkfs , listShares
Oct 17 18:21:42 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:21:46 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:21:50 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:21:54 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:21:58 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:02 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:06 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:10 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:14 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:18 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:22 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:26 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:30 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:34 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:38 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Oct 17 18:22:42 volumio volumio[904]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
......

xipmix, 2016-11-30:
This is an odd one. The problem seems to be with 'sudo', than with anything in the code.

> Oct 17 18:21:38 volumio sudo[1592]: pam_unix(sudo:auth): conversation failed
> Oct 17 18:21:38 volumio sudo[1592]: pam_unix(sudo:auth): auth could not identify password for [volumio]

If you try again with a freshly-flashed 2.031 does it still fail? If it does can you login (volumio@volumio) and paste the results of these commands:
```
$ id
$ sudo -l
```

RastaX, 2016-11-30:
I'll try the new version for Ordroid C2 (2.030) and report back until weekend!
Thanks for your input!

RastaX, 2016-12-01:
Same behaviour on version 2.030, here's the output:

`Dec 01 20:21:28 volumio volumio[1693]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Dec 01 20:21:32 volumio volumio[1693]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Dec 01 20:21:34 volumio volumio[1693]: info: CoreCommandRouter::executeOnPlugin: networkfs , editShare
Dec 01 20:21:34 volumio volumio[1693]: { path: 'elements2/_templib',
Dec 01 20:21:34 volumio volumio[1693]: ip: 'NAS',
Dec 01 20:21:34 volumio volumio[1693]: id: 'SCRAMBLED_ID_FIGURES',
Dec 01 20:21:34 volumio volumio[1693]: name: 'elements2',
Dec 01 20:21:34 volumio volumio[1693]: fstype: 'cifs',
Dec 01 20:21:34 volumio volumio[1693]: username: 'MYUSERNAME',
Dec 01 20:21:34 volumio volumio[1693]: password: 'MYPASSWORD',
Dec 01 20:21:34 volumio volumio[1693]: options: '',
Dec 01 20:21:34 volumio volumio[1693]: mounted: true,
Dec 01 20:21:34 volumio volumio[1693]: size: '1.11 TB' }
Dec 01 20:21:34 volumio sudo[1876]: pam_unix(sudo:auth): conversation failed
Dec 01 20:21:34 volumio sudo[1876]: pam_unix(sudo:auth): auth could not identify password for [volumio]
Dec 01 20:21:35 volumio volumio[1693]: info: CoreCommandRouter::executeOnPlugin: networkfs , listShares
Dec 01 20:21:36 volumio volumio[1693]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Dec 01 20:21:40 volumio volumio[1693]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Dec 01 20:21:44 volumio volumio[1693]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Dec 01 20:21:48 volumio volumio[1693]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
`

RastaX, 2016-12-01:
volumio@volumio:~$ id
uid=1000(volumio) gid=1000(volumio) groups=1000(volumio),4(adm),20(dialout),24(cdrom),25(floppy),29(audio),30(dip),44(video),46(plugdev),102(netdev)
volumio@volumio:~$ sudo -l
Matching Defaults entries for volumio on localhost:
    env_reset, mail_badpass, secure_path=/usr/local/sbin\:/usr/local/bin\:/usr/sbin\:/usr/bin\:/sbin\:/bin

User volumio may run the following commands on localhost:
    (ALL) ALL
    (ALL) NOPASSWD: /sbin/poweroff, /sbin/shutdown, /sbin/reboot, /sbin/halt, /bin/systemctl, /usr/bin/apt-get, /usr/sbin/update-rc.d, /usr/bin/gpio, /bin/mount, /bin/umount/, /sbin/iwconfig,
        /sbin/iwlist, /sbin/ifconfig, /usr/bin/killall, /bin/ip, /usr/sbin/service, /etc/init.d/netplug, /bin/journalctl, /bin/chmod, /sbin/ethtool, /usr/sbin/alsactl, /bin/tar, /usr/bin/dtoverlay,
        /sbin/dhclient, /usr/sbin/i2cdetect, /sbin/dhcpcd, /usr/bin/alsactl, /bin/mv, /sbin/iw, /bin/hostname
volumio@volumio:~$

RastaX, 2016-12-01:
screenshot before changing the path in the input-box:

[image]

screenshot after changing the path in the input-box but before clicking "SAVE":

[image]

After clicking SAVE the path stays "...templib" for a few seconds, then magically reverts back to "...musiklib"

RastaX, 2016-12-01:
I'll be glad to test and/or answer questions about the setup (same as in the first post)...
I flashed the Image-File to emmc without any additional changes to the system, other than over the webinterface.

xipmix, 2016-12-03:
Do you actually use the 'NAS' name in the IP address field, or is that just to hide the real address?
If you are using a name there you should check that the name resolves correctly:
```
$ ssh volumio@volumio
volumio$ host NAS
# should show an IP address
```
The next thing to test is running the command manually (if you use 'NAS', try that and try IP address)
```
volumio$ mkdir /tmp/test

# either
volumio$ /usr/bin/sudo /bin/mount -t cifs -o guest,ro,dir_mode=0777,file_mode=0666,iocharset=utf8,noauto //NAS/elements2/_templib /tmp/test
# or
volumio$ /usr/bin/sudo /bin/mount -t cifs -o guest,ro,dir_mode=0777,file_mode=0666,iocharset=utf8,noauto //<ipaddress>/elements2/_templib /tmp/test

volumio$ df -Ph /tmp/test; ls /tmp/test
volumio$ /usr/bin/sudo umount /tmp/test
```
sudo should not prompt you for a password when you try to mount the share. That seems to be what it is complaining about in the log; that it wants to ask for a password but none gets provided.
I checked your sudo -l output against what I get and it looks identical. So it looks like sudo is correctly configured. Possibly it is PAM.

xipmix, 2016-12-03:
Silly me. I should have tried to reproduce this and I have found that I can. That will help me dig deeper.

xipmix, 2016-12-03:
Got it. Look closely at these two commands that volumio is allowed to run.
```
.../bin/mount, /bin/umount/, ...
```
What do you notice?

xipmix, 2016-12-03:
I think [this PR](https://github.com/volumio/Build/pull/110) will fix it.

RastaX, 2016-12-03:
Let's get rid of the slash ;)
Thanks, I'm glad you found it!

closed 2016-12-03 by RastaX