volumio/Volumio2 #780: Is deactivating the WiFi deactivating the wired network as well ?

NicoPL, 2016-10-18:
Hello,
I found that on the Raspberry 3 which is wire connected, when deactivating the WiFi then restarting Volumio, the raspberry doesn't appear on the local network anymore.
Impossible to see it with a discover, access it by web (http://xxx.xxx.xxx.xxx, or http://volumio), neither by ssh (ssh pi@xxx.xxx.xxx.xxx).
Regards,

volumio, 2016-10-18:
How did you deactivate wifi?

NicoPL, 2016-10-19:
My mistake, it was not deactivating the wifi, but deactivating the "IP
Automatic" of the wifi.
Therefore : When deactivating the "IP Automatic" of the wifi, even the
wired network does'nt seem to work anymore.
And I can add a new question: Is it possible to deactivate the wifi (for a
wired rapsberry3 for example) within volumio ?

xipmix, 2016-10-19:
I started to try to understand this, using an rpi2 with wireless dongle fitted. (I don't think it matters if it is removable or not, just that both are activated at the start). I got quite different results, which I'll show in the next post.

It would be interesting if you could post what version you are running and logs of your test.

Question: what are you mainly after here - being able to turn off the wireless interface entirely (because you don't want to use it) or do you just want to set a static IP address for that interface?

NicoPL, 2016-10-19:
I tried it with the 0.979 and the 2.001 releases, which are installed on a
SD card on a Raspberry 3.

Actually I don't have any log (can't access them), because as soon as I
switch the "Automatic IP" off and restart the Raspberry, I can't access it
anymore with ssh. And reflash the SD card with a Volumio release. I don't
have spare mouse, keyboard and screen to connect to the Raspberry.

What kind of log file would you need ? I'll try to get spare peripherals.

What I wish is turning off the wireless interface of the Raspberry because
I don't need it.

volumio, 2016-10-19:
So, let me understand, you're switching off the wireless by selecting a static IP (do you put actual values in the static fields, if yes are they correct and appropriate?)

xipmix, 2016-10-19:
The test I did was with 2.000 on rpi2 with asus nano usb wifi (8192cu kernel module)

I start with both wired and wireless interfaces activated.
When I set the wifi auto-ip slider to Off, the 'static ip'/netmask/gateway fields appear and show values in them that I never entered (192.168.1.x network). They are in grey and I cannot edit them away, so I ignore them.
                                                                                
After pressing Save and reloading the web page, these fields all change to 'undefined'.
After Save is pressed, the system restarts _both_ the wired and wireless interfaces (see below).
                                                                                
After reloading the web ui, it still shows the wifi AutoIP as off, with the 'static' fields undefined.
But the status display at the top of the page shows wifi is still connected, and ifconfig and iwconfig confirm this.
I checked in /data/configuration/system_controller/network/config.json. This shows some expected things
```                                                                             
  "wirelessdhcp": {                                                             
    "type": "boolean",                                                          
    "value": false                                                              
  },                                                                            
  "wirelessip": {                                                               
    "type": "string",                                                           
    "value": "undefined"                                                        
  },                                                                            
  "wirelessnetmask": {                                                          
    "type": "string",                                                           
    "value": "undefined"                                                        
  },                                                                            
  "wirelessgateway": {                                                          
    "type": "string",                                                           
    "value": "undefined"                                                        
  }                                                                             
```                                                                             
and some  quite surprising things.                                              
```                                                                             
  "ethip": {                                                                    
    "value": "127.0.0.1",                                                       
    "type": "string"                                                            
  },                                                                            
  "ethnetmask": {                                                               
    "value": "255.255.255.0",                                                   
    "type": "string"                                                            
  },                                                                            
  "ethgateway": {                                                               
    "value": "0.0.0.0",                                                         
    "type": "string"                                                            
  },                                                                            
```                                                                             
After another webui reload, the status box for the ethernet connection goes away, which is consistent with the config.js file. But the ethernet interface is up and working just fine, which looks like a bug in the way the status data is being saved.

Running 'journalctl -f' while doing this experiment I noticed that it takes down both eth0 and wlan0 and brings them up again.
```                                                                             
Oct 19 07:32:55 volumio volumio[956]: info: CALLMETHOD: system_controller network saveWirelessNet [object Object]
Oct 19 07:32:55 volumio volumio[956]: info: CoreCommandRouter::executeOnPlugin: network , saveWirelessNet
Oct 19 07:32:55 volumio sudo[1202]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/chmod 777 /etc/network/interfaces
Oct 19 07:32:55 volumio sudo[1202]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 19 07:32:55 volumio sudo[1202]: pam_unix(sudo:session): session closed for user root
Oct 19 07:32:55 volumio sudo[1210]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/chmod 777 /etc/dhcpcd.conf
Oct 19 07:32:55 volumio sudo[1210]: pam_unix(sudo:session): session opened for user root by (uid=0)
Oct 19 07:32:55 volumio sudo[1210]: pam_unix(sudo:session): session closed for user root
Oct 19 07:32:55 volumio volumio[956]: info: Permissions for /etc/network/interfaces set
Oct 19 07:32:55 volumio sudo[1220]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl restart wireless.service
Oct 19 07:32:55 volumio sudo[1221]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/ip addr flush dev eth0
Oct 19 07:32:55 volumio systemd[1]: Stopping Wireless Services...               
[... cut ...]
Oct 19 07:33:22 volumio systemd[1]: Started UPnP Renderer front-end to MPD.     
Oct 19 07:33:22 volumio sudo[1578]: pam_unix(sudo:session): session closed for user root
Oct 19 07:33:22 volumio volumio[956]: info: Upmpdcli Daemon Started             
Oct 19 07:33:22 volumio mpd[560]: client: [3] opened from ::ffff:127.0.0.1:59722
```
If I then reboot the system, it starts with eth0 up and wlan0 down, but after a little while (<30s) the wireless interface comes up again. In the webui, only the wireless interfaces is shown as active.

xipmix, 2016-10-19:
What I am wondering at this point is whether the wireless.js calls to dhcpcd need to specify the interface (wlan0) explicitly, so that eth0 is left alone during this process. There is another process related to eth0 running, ie (after rebooting)
```
$ ps -fade|grep dhcpc
root       707     1  0 12:11 ?        00:00:00 dhcpcd eth0
root       869     1  0 12:11 ?        00:00:00 /sbin/dhcpcd
```
so I may well be on the wrong track here.

I also wonder if systemd needs a bit more guidance with restarting wireless.service, i.e. would adding
```
ExecStop=/volumio/app/plugins/system_controller/network/wireless.js stop
```
to the unit file help at all. At present it relies on ```KillMode=mixed``` knocking off the right group of processes and if the dhcp client process for eth0 is one of those...

NicoPL, 2016-10-20:
To answer to "Volumio", I tried to disable the wifi. And actually did not
do it but switch off the Automatic IP of the WiFi only.
But the drawback is that when restarting the raspberry, it doesn't appear
on the local network anymore even if connected with a wire.
I did not change anything about the parameters of wifi and ethernet, except
switching off the "Automatic IP" of the Wifi.

closed 2017-06-02 by volumio