volumio/Volumio2 #788: Mixer Type does not work

Bergmann1, 2016-10-20:
Hello,

In Volumio 2.001 on RB2+ Mixer Type menu does not work well, the only selection that I am getting is "Hardware". I tried to select the Mixer Type with Output Device "Audio Jack", "HDMI" and "DAC", but nothing has changed...the Mixer Type only gives the option for "Hardware".

balbuze, 2016-10-23:
what is the problem for you of having "hardware" mixer ? It is porposed because your dac supports it! So don't need a software mixer.

Bergmann1, 2016-10-23:
I do not have any problems with having the hardware mixer, but it does not work. When I increase or decrease the volume by volume bar, nothing happens, the sound volume is constant.

Bergmann1, 2016-10-23:
I had the same problem in 1.55, so I was using the software mixer instead the hardware one.

balbuze, 2016-10-23:
what is your dac ?

Bergmann1, 2016-10-23:
HRT Music Streamer II

balbuze, 2016-10-23:
I see... USB dac. I 've got the same problem with a usb FDA (fx audio 802). Alsa gives a pcm mixer with no effect... I made a test with Ubuntu with the same behaviour. It seems something in wrong in alsa...

Bergmann1, 2016-10-23:
Strange...did 1.55 use also alsa?

balbuze, 2016-10-23:
Yes but when choosing software mixer in 1.55 it was managed by mpd. The problem was with other app such as -spop (for Spotify) that don't use mpd and then no mixer

closed 2017-06-03 by volumio