volumio/Volumio2 #793: Deleting webradio item deletes all with same URL

jo-me, 2016-10-21:
when I'm in "Browse" -> "Web Radio" -> "Favorite Radios" and delete an entry there via "Remove from Radio favorites" any webradio station with the same URL is deleted along with the selected entry.

So in my case I accidentally added a station twice which was created with a different name (in that case a name including the currently playing song). When I deleted the duplicate entry, the original entry disappeared with it. So then I had to re-add the station to the favorites.

The associated forum thread is here: https://volumio.org/forum/issue-when-deleting-webradio-favorite-t4846.html

jo-me, 2016-10-21:
PlaylistManager.prototype.commonRemoveFromPlaylist() checks only equality of URI and serviceName ("webradio" in this case) when deleting stations.

volumio, 2016-10-21:
We should send also the name to this function, which will probably break something ...
Unfortunately this will go NOFIX...

closed 2016-10-21 by volumio