volumio/Volumio2 #795: Volumio as Bluetooth Audio Receiver

escoand, 2016-10-21:
It would be nice to use volumio as a bluetooth audio receiver.
So it could additionally be used for receiving audio streams from smartphone or pc as central audio "device".

I've read about in a german magazine (not tested by myself) if this will help:
https://shop.heise.de/katalog/raspberry-in-blue

karaambaa, 2016-10-30:
Don't know if it helps, but you could take a short look at these instructions on to install a2dp-bluetooth on raspbian jessy [https://github.com/karaambaa/a2dp-bluetooth-server](https://github.com/karaambaa/a2dp-bluetooth-server)

AbysFlo, 2017-03-08:
Got it up and running with the following scripts, using only the bluetooth scripts

https://bitbucket.org/adenbeckitt/raspberry-pi-audio-receiver-install/src/93032f81ec2a338a28473d1c067448fbb35318aa?at=master

Simtwo, 2017-06-29:
I can't seem to find how to make bluetooth work properly. Does someone work on a plugin ? 
It would be nice !

xipmix, 2017-06-30:
You may also be interested in @wvengen 's approach, see  #159

death2all110, 2017-08-13:
@AbysFlo can you explain which scripts you ran exactly?
Also, does pulse interfere with Volumio2 at all? I understand Volumio2 uses ALSA instead of Pulse.

AbysFlo, 2017-08-21:
@death2all110 If I recall well I ran bt_pa_config.sh and bt_pa_install.sh and that was up and running. Not sure how they mixed together, like might needed to pause a song to have only one source of music at the time. I didn't pursue my testing as I have a phone running on 2.4ghz wifi and bluetooth was interfering with it.. I think pulse crashed couple of time as well that needed a reboot. But was usable if you're not stuck with 2.4ghz wifi

donjuan201, 2017-09-13:
I think a very cool and super awesome feature would be if you could stream from a smartphone (through a Bluetooth or Wifi (maybe through the app as well) connection) to volumio and have volumio record the audio input and give you the option to save it to your local / network device containing your media files. This way you can also expand your library if anyone comes past and has a song that you don't have in your library, you can then easily have them stream it through your volumio setup and boom you have the song now part of your library...

Make it sound easy, right, I think a lot of people will find this feature super handy.

Zjerardo, 2017-10-25:
I have also tried to install this using the https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install method, but it still isn't working.

steps I've done so far:
sudo apt-get install build-essential
sudo apt-get install libtool
sudo apt-get install intltool
sudo apt-get install locales
sudo dpkg-reconfigure locales

afterwards I run the installer.sh (sudo) and select option 4 (bluetooth only).

it seems to run through the installation process, but it doesn't activate the bluetooth.
I'll have to recheck for any other error messages in the log and try to resolve them but if anyone else has a proper installation procedure to get this working, it would be greatly appreciated!

jaglo, 2017-11-10:
I had the same issue with the above script.  Looking over my install.log file, it looked like the script was aborting any time it was prompted to install more than 1 package at a time.  I was able to successfully get Bluetooth working on my Volumio installation by installing all the prerequisites first and then running the script again:

`sudo apt-get install autoconf automake autopoint autotools-dev file gettext gettext-base libasprintf-dev libasprintf0c2 libauthen-sasl-perl libcroco3 libencode-locale-perl libfile-listing-perl libfont-afm-perl libgettextpo-dev libgettextpo0 libhtml-form-perl libhtml-format-perl libhtml-parser-perl libhtml-tagset-perl libhtml-tree-perl libhttp-cookies-perl libhttp-daemon-perl libhttp-date-perl libhttp-message-perl libhttp-negotiate-perl libio-html-perl libio-socket-ssl-perl liblwp-mediatypes-perl liblwp-protocol-https-perl libmailtools-perl libnet-http-perl libnet-smtp-ssl-perl libnet-ssleay-perl libsigsegv2 libtimedate-perl libunistring0 liburi-perl libwww-perl libwww-robotrules-perl libxml-parser-perl m4 patch libflac-dev libogg-dev libsndfile1-dev libvorbis-dev libjson-c-dev libjson0 libjson0-dev make`

kerwingroot, 2017-12-12:
Hi, i'm still having some problems with installing the plugin.
I'm connected trough ssh and the installer seems to be working ok, but suddenly ends with this and closes connection:

```
===> Executing: su volumio -c ./bt_pa_config.sh
mkdir: cannot create directory '/home/pi/pyScripts': No such file or directory
===> Executing: sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
sudo: no tty present and no askpass program specified
Exiting script due to error from: sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
Exiting script due to error from: su volumio -c ./bt_pa_config.sh
--------------------------------------------
./install.sh: line 291: [: !=: unary operator expected
Ending at @ Tue Dec 12 12:48:59 UTC 2017
 
```

If anybody has some great ideas to work around this, that would be great!

xipmix, 2017-12-13:
This is an issue that should be reported to the author of bt_pa_config.sh.
There are lots of problems here.
 * The script is hardcoding the /home/pi/ directory which doesn't exist on your system
 * The sudo calls are not handled correctly; bt_pa_config.sh should be run via sudo
 * The install.sh script needs to use proper quoting and seems to have a logic flaw.

biva, 2017-12-13:
For information, there's a discussion on the forum: https://volumio.org/forum/a2dp-bluetooth-support-t1475-40.html

And some improvements have been done by @BaReinhard here: https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install

biva, 2017-12-13:
See also https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/76 and https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/issues/71#issuecomment-350416641

BaReinhard, 2017-12-13:
I have updated the repo with a `volumio_fix` branch to potentially fix several issues. However, I dont have access to volumio so I can't test at the moment. 

@xipmix I am open to making changes and very much open to pull requests with potential fixes you have in mind. I potentially have some free time this coming week to setup a volumio pi and testing things out.

BaReinhard, 2017-12-23:
@xipmix thanks for some of your efforts to fix the install script. I have taken some time and cleaned up most if not all of the code. I believe it fixed most of the issues. I have removed all of the pi only user problems (that I’m aware of). That being said in the coming weeks I am looking to debug with other installs, I just recently got snapcast support so now I’m going to be looking at volumio as it seems like it’s the next big use case for many users. Not sure How inclusive it will be but I’d like for it to work for the average volumio user.

biva, 2017-12-28:
@BaReinhard great news, that you're looking into Volumio :) Thank you.
Are you able to test it? I'm available to test it if you want (RPI 3)

BaReinhard, 2017-12-28:
@biva I haven't gotten a chance yet, but i'll setup a branch for other users to test with as well. I am assuming you're more interested in bluetooth than anything else?

biva, 2017-12-28:
Yes, bluetooth is my main (probably the single) interest. Thank you again!

harel-e, 2017-12-29:
+1 on bluetooth

kekoa428, 2018-01-03:
+1 bluetooth with volumio.

BaReinhard, 2018-01-04:
Good news ladies and gentlemen. I have it working for volumio. A few issues to iron out but should be done by tomorrow.

harel-e, 2018-01-04:
@BaReinhard, thanks a lot for the update.
Following your instructions, I was able to stream via Bluetooth. Awesome !
I'm using Raspberry Pi 3

I noticed that AirPlay got disabled. Is there a way to have both?

Thanks again

wvengen, 2018-01-04:
Nice! I hope this can eventually get integrated into Volumio. Looking into the scripts, the steps for bluetooth don't look too complicated :)

BaReinhard, 2018-01-04:
@wvengen The scripts for Bluetooth aren’t that difficult especially if they are shipping a baked image it should be easier to do. It’s just integration with other features that will need the tests.

@harel-e what do you mean it got disabled? Does volumio come with airplay installed?

Zjerardo, 2018-01-04:
@BaReinhard: volumio offers Airplay functionality out of the box I think.

I'll be able to reïnstall my volumio system over the weekend and test out your BT functionality! Thanks!

BaReinhard, 2018-01-04:
@harel-e @Zjerardo Thanks, and yes I just checked and airplay functionality is intact with this installation, the metadata shows up on volumio.local . Bluetooth may take more of a chore to get working with volumio. I may not have the spare time for something like this, but I'll be taking a look at it anyway.

BaReinhard, 2018-01-08:
If there is someone who is familiar with creating Volumio plugin's and would like to collaborate to make this functionality easily added, instead of needing to ssh and run commands in the CLI. Please reach out I would love to make this easier for the end user.

matrix321321, 2018-01-13:
@BaReinhard: Thank you !

snowf0x, 2018-01-15:
I confirm BaReinhard's instruction works well for me. I'm on Rpi3 as well. Thanks a lot!

BaReinhard, 2018-01-15:
Those using my repo are any of you experiencing any choppy playback?

biva, 2018-01-15:
@BaReinhard Yes, choppy and delayed :(

BaReinhard, 2018-01-15:
Is anyone hearing clear playback?

SonosKiller, 2018-01-15:
Hi @BaReinhard and thanks for you work

Using a RPI3 and Jack and Hdmi as output

Installed mid last week by SSH and had choppy sound from both Pc and phone. Disbled the wifi and used cable instead and then the sound was good from both

Yesterday I did a clean install and a upgrade to current version , installed via SSH , changed the "audio_sink=0" and the sound from the Pc is crisp and clear but from the phone it's still choppy 
(Currently using Wifi , have't had the time to test with cable yet)

BaReinhard, 2018-01-15:
On WiFi bluetooth streaming will always be choppy. Unless you use an external Bluetooth or WiFi adapter. The issue is the Broadcom chip:
https://github.com/raspberrypi/linux/issues/1402

Simtwo, 2018-01-16:
Glad to see that it can work. I am personnally really interested in a plugin to test it !!

BaReinhard, 2018-01-16:
Plugin needs work, current one was pretty hacky. I would still suggest using the ssh install. I don’t think I’ll have much time for a plugin just yet. That being said I’m still looking into options of making this easier without all the extra complicated dependencies.

SonosKiller, 2018-01-16:
Hi @BaReinhard

I'm aware off the wifi / bluetooth issue 
Just by curiosity, any idea why it sudently works flawless from the Pc ?
Just lucky or a medical wonder ?  :)

xipmix, 2018-01-17:
To disable the inbuilt wifi you may have to blacklist the kernel module it needs. There should be examples & helpers over in the raspberrypi.org forums.

The dongle - try plugging in, then run 'dmesg' to see if the kernel noticed, and 'lsusb' to see if it was configured.

macmpi, 2018-01-17:
FWIW, an easy & proper way to disable onboard wifi on Pi 3 / PiZeroW is to simply add:
[`dtoverlay=pi3-disable-wifi`](https://github.com/Hexxeh/rpi-firmware/tree/master/overlays) statement within `/boot/config.txt`

SonosKiller, 2018-01-17:
@macmpi Thank you ! :)
Works nice and smooth now

macmpi, 2018-01-25:
also note that `bluez-firmware` ( **1.2-3+rpt1** or later) has been recently improved to overcome some wifi interrop issues...
You may check if `sudo apt-get update; sudo apt-get install bluez-firmware` pulls a newer version.

fantaaaaaa, 2018-01-25:
just came to the idea to try volumio with bluetooth and stumbled upon this thread. great to see that theres some active development. 

can you please just clear this up for me since i am not that advanced on unix yet :) can i run my ssh'd volumio install (installed from the available volumio-image) with airplay and bluetooth by installing the "Bluetooth Only" option?

@BaReinhard please keep us posted on the plugin. The effort is very much appreciated!!

can anybody report if the firmware mentioned by @macmpi improves the choppy playback? i use my raspberry in a mobile speaker setup and would love to keep wifi enabled for airplay since i use this most often. but bluetooth would be great if you use the speaker on the road :)

BaReinhard, 2018-01-28:
As much as I’d like to continue development as frequently as I had been in the past few months, my new jobs workload has considerably ate into a lot of my open source development time. That being said, I’ll try to continue with some support, but it looks that for the moment plug-in development has stopped. I may touch back on it later, but not for a few months. If anyone is interested in doing the development for it I will more than happily contribute in anyway that I can. @fantaaaaaa as far as keeping WiFi and bluetooth, an external WiFi card will fix the choppy playback issues and is a pretty cheap work around.

fantaaaaaa, 2018-01-29:
@BaReinhard thanks for your feedback and your effort. Just to be clear: if i install the Bluetooth Only option on my Volumio Image, i can use both and when using bluetooth it will disable my wifi for that time?

Thanks!

BaReinhard, 2018-01-29:
Yes you can use both with the bluetooth only install, at least from my experience with it. If it does affect AirPlay, it would only require a minor change. Currently, wifi is still enabled during bluetooth playback, but there is an option to turn off wifi when bluetooth is connected, I will have to look where that script is located.

alfieoriginal, 2018-02-02:
@BaReinhard  thanks a lot for your work.
I guess that regardless of this feature is shown in the project page
> Ability to disable Wifi on Bluetooth Connection. Needs to be enabled, but solves choppy bluetooth playback if no external wifi card is available.
Still not completely working. Isn't it?

BaReinhard, 2018-02-02:
It should work but needs to be enabled. I haven’t had a chance to debug it if errors do occur. It worked on a few reported occasions but I didn’t ask more in depth questions. I’ll mark I as a todo item, and provide some documentation

nectronx, 2018-02-03:
hello everyone that's very good. Awesome !!!
 I tried it and it worked.
but I need to know 2 things guys.
1111-  if anyway we can use the bluetooth as output audio.
SCENARIO :  RPI3 with Volumio using a bluetooth speaker. no jack in the speaker. I want to connect my RPI3_Volumio in that speaker via bluetooth. 
I would to know if I can find and plugin.

2222- Do anyone know if i can find a AutoFade audio plugin for volumio.

thank you

BaReinhard, 2018-02-05:
What are you looking for as far as auto fade? Slowly increasing audio after a bt connection?

Bluetooth connection to Bluetooth speaker is pretty easy. Take a look at this repo I created, I haven’t tested on volumio and I don’t know that the install script would work for it, but you can follow the code and do it manually. https://github.com/BaReinhard/a2dp_bluetooth

SonosKiller, 2018-02-18:
Hi trying to reinstall , but it don't work as last time !

> git clone https://github.com/bareinhard/super-simple-raspberry-pi-audio-receiver-instal

Asks for a username and password

> Username for 'https://github.com':

BaReinhard, 2018-03-03:
I would ensure that the path is correct: `git clone https://github.com/bareinhard/super-simple-raspberry-pi-audio-receiver-install` (with 2 l at the end) I haven't had this issue before so I can't say for sure, but this is a git issue. (Possibly some config issues?)

plomkiolki, 2018-03-09:
hello and thanks for the script.
i ve tested on a raspberry pi 3 after a fresh install of volumio and unfortunately i get an error after running the script:
SSRPARI [Fri Mar  9 21:13:00 UTC 2018] Executing sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
Fatal error encountered: 'sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py' failed
Fatal error encountered: su volumio -c ./bt_pa_config.sh failed

any idea?

twoolums, 2018-03-13:
Not sure what I'm doing wrong but I'm getting allot of packages not found from a stock volumio image.  

SSRPARI [Tue Mar 13 13:23:18 UTC 2018] Restoring Original Files...
.profile
SSRPARI [Tue Mar 13 13:23:18 UTC 2018] Saving /home/volumio/.profile...
.profile=/home/volumio/
SSRPARI [Tue Mar 13 13:23:18 UTC 2018] Select Your Install Options
SSRPARI 1. Install the Raspberry Pi Audio Receiver Car Installation
SSRPARI 2. Install the Raspberry Pi Audio Receiver Home Installation
SSRPARI 3. Install the Raspberry Pi Network Without Internet Installation (For teaching!)
SSRPARI 4. Install the Bluetooth Only Installation
SSRPARI 5. Install the Snapcast Installation (BETA), choose from Snapcast Server, Client, or Both (Requires Minor Configuration)
SSRPARI 6. Install a Custom Raspberry Pi Audio Receiver
Which installation would you like to choose? (1/2/3/4/5/6) : 4
Do you want all the Devices to use the same name? (y/n) : y
Device name: GarageMusic
SSRPARI [Tue Mar 13 13:23:34 UTC 2018] Updating via Apt-Get
SSRPARI [Tue Mar 13 13:24:09 UTC 2018] Upgrading via Apt-Get
SSRPARI [Tue Mar 13 13:24:11 UTC 2018] Running ./bt_pa_install.sh...
SSRPARI [Tue Mar 13 13:24:11 UTC 2018] Checking pulseaudio-module-bluetooth...
dpkg-query: no packages found matching pulseaudio-module-bluetooth
SSRPARI [Tue Mar 13 13:24:11 UTC 2018] Installing pulseaudio-module-bluetooth...
SSRPARI [Tue Mar 13 13:24:32 UTC 2018] Checking python-dbus...
SSRPARI [Tue Mar 13 13:24:32 UTC 2018] Dependency python-dbus already met...
SSRPARI [Tue Mar 13 13:24:32 UTC 2018] Checking libltdl-dev...
dpkg-query: no packages found matching libltdl-dev
SSRPARI [Tue Mar 13 13:24:32 UTC 2018] Installing libltdl-dev...
SSRPARI [Tue Mar 13 13:25:10 UTC 2018] Checking pulseaudio...
SSRPARI [Tue Mar 13 13:25:10 UTC 2018] Dependency pulseaudio already met...
SSRPARI [Tue Mar 13 13:25:10 UTC 2018] Checking libtool...
SSRPARI [Tue Mar 13 13:25:10 UTC 2018] Dependency libtool already met...
SSRPARI [Tue Mar 13 13:25:10 UTC 2018] Checking intltool...
dpkg-query: no packages found matching intltool
SSRPARI [Tue Mar 13 13:25:10 UTC 2018] Installing intltool...
SSRPARI [Tue Mar 13 13:25:39 UTC 2018] Checking libsndfile-dev...
dpkg-query: no packages found matching libsndfile-dev
SSRPARI [Tue Mar 13 13:25:39 UTC 2018] Installing libsndfile-dev...
SSRPARI [Tue Mar 13 13:25:49 UTC 2018] Checking libcap-dev...
dpkg-query: no packages found matching libcap-dev
SSRPARI [Tue Mar 13 13:25:49 UTC 2018] Installing libcap-dev...
SSRPARI [Tue Mar 13 13:25:55 UTC 2018] Checking libjson0-dev...
dpkg-query: no packages found matching libjson0-dev
SSRPARI [Tue Mar 13 13:25:56 UTC 2018] Installing libjson0-dev...
SSRPARI [Tue Mar 13 13:26:03 UTC 2018] Checking libasound2-dev...
SSRPARI [Tue Mar 13 13:26:03 UTC 2018] Dependency libasound2-dev already met...
SSRPARI [Tue Mar 13 13:26:03 UTC 2018] Checking libavahi-client-dev...
SSRPARI [Tue Mar 13 13:26:03 UTC 2018] Dependency libavahi-client-dev already met...
SSRPARI [Tue Mar 13 13:26:03 UTC 2018] Checking libbluetooth-dev...
dpkg-query: no packages found matching libbluetooth-dev
SSRPARI [Tue Mar 13 13:26:03 UTC 2018] Installing libbluetooth-dev...
SSRPARI [Tue Mar 13 13:26:10 UTC 2018] Checking libglib2.0-dev...
SSRPARI [Tue Mar 13 13:26:11 UTC 2018] Installing libglib2.0-dev...
SSRPARI [Tue Mar 13 13:26:24 UTC 2018] Checking libsamplerate0-dev...
dpkg-query: no packages found matching libsamplerate0-dev
SSRPARI [Tue Mar 13 13:26:24 UTC 2018] Installing libsamplerate0-dev...
SSRPARI [Tue Mar 13 13:26:32 UTC 2018] Checking libsbc-dev...
dpkg-query: no packages found matching libsbc-dev
SSRPARI [Tue Mar 13 13:26:32 UTC 2018] Installing libsbc-dev...
SSRPARI [Tue Mar 13 13:26:37 UTC 2018] Checking libspeexdsp-dev...
dpkg-query: no packages found matching libspeexdsp-dev
SSRPARI [Tue Mar 13 13:26:37 UTC 2018] Installing libspeexdsp-dev...
SSRPARI [Tue Mar 13 13:26:45 UTC 2018] Checking libssl-dev...
dpkg-query: no packages found matching libssl-dev
SSRPARI [Tue Mar 13 13:26:45 UTC 2018] Installing libssl-dev...
SSRPARI [Tue Mar 13 13:26:54 UTC 2018] Checking libtdb-dev...
dpkg-query: no packages found matching libtdb-dev
SSRPARI [Tue Mar 13 13:26:54 UTC 2018] Installing libtdb-dev...
SSRPARI [Tue Mar 13 13:27:00 UTC 2018] Checking libbluetooth-dev...
SSRPARI [Tue Mar 13 13:27:00 UTC 2018] Dependency libbluetooth-dev already met...
SSRPARI [Tue Mar 13 13:27:00 UTC 2018] Checking intltool...
[... 69 lines cut ...]
SSRPARI [Tue Mar 13 13:28:30 UTC 2018] Checking libasound2-dev...
SSRPARI SSRPARI [Tue Mar 13 13:28:30 UTC 2018] Dependency libasound2-dev already met...
[Tue Mar 13 13:28:30 UTC 2018] Checking libavahi-client-dev...
SSRPARI SSRPARI [Tue Mar 13 13:28:30 UTC 2018] Dependency libavahi-client-dev already met...
[Tue Mar 13 13:28:30 UTC 2018] Checking libspeexdsp-dev...
SSRPARI [Tue Mar 13 13:28:30 UTC 2018] Dependency libspeexdsp-dev already met...
SSRPARI [Tue Mar 13 13:28:30 UTC 2018] Checking liborc-0.4-dev...
dpkg-query: no packages found matching liborc-0.4-dev
SSRPARI [Tue Mar 13 13:28:30 UTC 2018] Installing liborc-0.4-dev...
SSRPARI [Tue Mar 13 13:28:36 UTC 2018] Checking intltool...
SSRPARI SSRPARI [Tue Mar 13 13:28:36 UTC 2018] Dependency intltool already met...
[Tue Mar 13 13:28:36 UTC 2018] Checking libtdb-dev...
SSRPARI [Tue Mar 13 13:28:36 UTC 2018] Dependency libtdb-dev already met...
SSRPARI [Tue Mar 13 13:28:36 UTC 2018] Checking libssl-dev...
[... cut ...]
SSRPARI [Tue Mar 13 13:28:36 UTC 2018] Checking libcap-dev...
SSRPARI [Tue Mar 13 13:28:36 UTC 2018] Dependency libcap-dev already met...
SSRPARI [Tue Mar 13 13:28:37 UTC 2018] Running su volumio -c ./bt_pa_config.sh...
SSRPARI [Tue Mar 13 13:28:37 UTC 2018] Executing sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py
Fatal error encountered: 'sudo cp usr/local/bin/volume-watcher.py /usr/local/bin/volume-watcher.py' failed
Fatal error encountered: su volumio -c ./bt_pa_config.sh failed

plomkiolki, 2018-04-11:
you need to use volumio version 2.348 to make it work.
it doesn t with the latest version.

wullxz, 2018-08-21:
The reason for the Fatal errors in bt_pa_config.sh is, that it is started as user 'volumio' which has no right to use sudo to become root without password. The exact error message, which gets swallowed by the `exc` function in functions.sh is: `sudo: no tty present and no askpass program specified`.

It is possible to temporarily solve that problem by adding the following line to `/etc/sudoers` and removing it after the installation ran successfully:
`volumio ALL=(ALL) NOPASSWD: ALL`

I thought about doing that automatically in `install.sh` but I'd rather know the reason to start bt_pa_config.sh as 'volumio' instead of root first @BaReinhard ?

wullxz, 2018-08-28:
I created a pull request for @BaReinhard's Super-Simple-Raspberry-Pi-Audio-Receiver-Install script to fix the sudo issue: https://github.com/BaReinhard/Super-Simple-Raspberry-Pi-Audio-Receiver-Install/pull/140

rototom, 2018-10-01:
I installed this plugin but cant find any configuration. What do I have to do after installing?

rototom, 2018-10-01:
I dont find any bluetooth devices...

SonosKiller, 2018-10-01:
are you using ver 2.348 ?

rototom, 2018-10-01:
2.457

[10 more replies]

closed 2020-01-13 by volumio