volumio/Volumio2 #799: IQAudio Digiamp+ unmute

eggerph, 2016-10-21:
Hi
Just wanted to report a small issue. I have a IQAudio Digimp+ and when I set the I2S DAC in the settings the first time and the system reboots the DigiAmp+ is also unmuted correctly. A reboot apparently still keeps pin 22 high and it is still unmuted. However when it is disconnected from the power source and booted up again the Digiamp+ is stays muted. I solved to problem currently by adding 
echo "22" > /sys/class/gpio/export
echo "out" >/sys/class/gpio/gpio22/direction
echo "1" >/sys/class/gpio/gpio22/value
to the rc.local file. However it would be nice if this would work without doing so.
cheers and thanks in advance!
Philipp

volumio, 2016-10-21:
Hi philipp I just solved this issue. Please try this image and tell me if that works:
http://updates.volumio.org/pi/volumio/2.004/volumio-2.004-2016-10-21-pi.img.zip

eggerph, 2016-10-24:
Hi, thanks for the quick reply. I tested the 2.004 and seems to work fine. Unplugged it once and Amp still worked after that. However the webUI has problems on my laptop. It doesn't load anymore. (attached a screenshot)
[image]
 Works fine on my tablet though. Is this known? 
But the actual issue seems to be solved!

closed 2016-10-24 by eggerph