volumio/Volumio2 #801: software volume control not working in 2.006

cortegedusage, 2016-10-23:
I've got a raspberry pi 3 with a cheap dac which is working with the generic i2s driver.
when using the software mixer. after a reboot it says it uses software mixer,
but there is no sound. only after saving this option again, it works.

besides from that the spotify connect plugin doesn't work with software mixer either.
when set to none. It works.

any ideas?

many thanks

balbuze, 2016-10-23:
The Spotify connect plugin is not finished yet. And your are right, with softvol there is no sound... I 'm still trying to make it work... What version of volumio ? What version of plugin ( you can see that in package.json in the plugin folder). With your dac if you use "none" as mixer you should get sound, but of course no volume control...

cortegedusage, 2016-10-24:
Hi Balbuze thank you for your reply, keep up the good work and I'm confident that you'll make it work.

as for the problem that the software volume is not persistent, as long as the spotify plugin doesn't support software volume i'll live with it. though it is not as expected.

many thanks

closed 2016-11-16 by volumio