volumio/Volumio2 #806: Option to use 'Album Artist' to populate artist list in WebUI

tpar7322, 2016-10-26:
I use MusicBrainz Picard to manage the metadata for my music library. For each track it has tags for 'Artists' and 'Album Artist'. Currently under the Artist menu the WebUI uses the 'Artists' tag to populate the list. That makes sense, but it results in lots of 'Artist feat ...' items in the list which makes navigation a bit more difficult. It would be nice to have the option to use the 'Album Artist' tag to populate the list rather than 'Artists' for those of us who do have both tags in the metadata.

mw767, 2016-10-27:
Same problem here, does anyone know which file to edit to list the albums by album artist to stop the listing of 40 albums for one various album? It was player_lib.php that needed editingin older versions, but cant find that file on Volumio 2 Odroid c2 . Any help much appreciated
Regards
Martin

tpar7322, 2016-11-02:
Issue #814 has been opened that covers the same request

closed 2017-06-02 by volumio