volumio/Volumio2 #810: volumio CLI not working for playback commands

jo-me, 2016-10-28:
The "volumio" command line interface throws errors on all commands that are handled by the "playback.js" file, so play / pause / next / etc.

Example:

volumio@volumio:~$ volumio play
Sending play
/volumio/app/plugins/system_controller/volumio_command_line_client/volumio.sh: /volumio/app/plugins/system_controller/volumio_command_line_client/commands/playback.js: /bin/node^M: bad interpreter: No such file or directory

The reason is in fact some invisible illegal character in the first line.

jgarriga78, 2016-11-23:
Hi,
open /volumio/app/plugins/system_controller/volumio_command_line_client/commands/playback.js in nano, save it with: ctrl+x -- YES -- alt+M to save it in Ms-Dos mode.

jo-me, 2016-11-23:
thanks, the fix is already in the current release.

closed 2016-11-23 by jo-me