volumio/Volumio2 #813: Spotify playlists stop after playing first track

skikirkwood, 2016-10-28:
This is odd, on 2.011 (and previously on 2.06) after starting a Spotify playlist it stops after the first track.  This is true all of all Spotify playlists, but all of the tracks of the playlist were successfully added to the play queue.

Tail end of the Volumio log below:

[1477685484557] ---------------------------- Client requests Volumio queue
2016-10-28T20:11:24.561Z - info: [1477685484560] InterfaceWebUI::pushQueue
2016-10-28T20:11:24.623Z - info: [1477685484622] ------------------------------ 71ms
2016-10-28T20:11:26.101Z - info: CoreCommandRouter::executeOnPlugin: spop , handleBrowseUri
2016-10-28T20:11:27.676Z - info: CoreCommandRouter::executeOnPlugin: spop , handleBrowseUri
2016-10-28T20:11:27.679Z - info: [1477685487678] ControllerSpop::sendSpopCommand
2016-10-28T20:11:27.680Z - info: ADDING DEFER FOR COMMAND ls
2016-10-28T20:11:27.697Z - info: FIRST BRANCH
2016-10-28T20:11:27.697Z - info: BEFORE: SPOP HAS 1 PROMISE IN STACK
2016-10-28T20:11:27.699Z - info: AFTER: SPOP HAS 0 PROMISE IN STACK
2016-10-28T20:11:32.780Z - info: CoreCommandRouter::volumioAddQueueItems
2016-10-28T20:11:32.781Z - info: [1477685492781] CoreStateMachine::addQueueItems
2016-10-28T20:11:32.783Z - info: [1477685492783] CorePlayQueue::addQueueItems
2016-10-28T20:11:32.786Z - info:  uri=spotify/playlists/1, title=Green Day – Revolution Radio, service=spop
2016-10-28T20:11:32.787Z - info: First index is 675
2016-10-28T20:11:32.788Z - info: ADDING THIS ITEM TO QUEUE: {"uri":"spotify/playlists/1","title":"Green Day – Revolution Radio","service":"spop"}
2016-10-28T20:11:32.789Z - info: Exploding uri spotify/playlists/1 in service spop
2016-10-28T20:11:32.792Z - info: [1477685492791] ControllerSpop::sendSpopCommand
2016-10-28T20:11:32.793Z - info: ADDING DEFER FOR COMMAND ls
2016-10-28T20:11:32.797Z - info: FIRST BRANCH
2016-10-28T20:11:32.797Z - info: BEFORE: SPOP HAS 1 PROMISE IN STACK
2016-10-28T20:11:32.798Z - info: AFTER: SPOP HAS 0 PROMISE IN STACK
2016-10-28T20:11:32.805Z - info: [1477685492804] CorePlayQueue::saveQueue
2016-10-28T20:11:32.880Z - info: Adding item to queue: [{"uri":"spotify:track:4UGWMzkmEPpYoS9myk8lAG","service":"spop","type":"song","name":"Somewhere Now","title":"Somewhere Now","artist":"Green Day","album":"Revolution Radio","duration":249,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:278Ao6KQDxWGGszv24uvhI","service":"spop","type":"song","name":"Bang Bang","title":"Bang Bang","artist":"Green Day","album":"Revolution Radio","duration":205,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:3oqYMRKQcwyNmFn1VD2ukD","service":"spop","type":"song","name":"Revolution Radio","title":"Revolution Radio","artist":"Green Day","album":"Revolution Radio","duration":181,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:2OV1oB6LQWNw7kTOHU7Bua","service":"spop","type":"song","name":"Say Goodbye","title":"Say Goodbye","artist":"Green Day","album":"Revolution Radio","duration":219,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:0PAlrNkCRRHv7hShF5s7Rr","service":"spop","type":"song","name":"Outlaws","title":"Outlaws","artist":"Green Day","album":"Revolution Radio","duration":302,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:2OEXuupzwtyc5kufQE6QLY","service":"spop","type":"song","name":"Bouncing Off the Wall","title":"Bouncing Off the Wall","artist":"Green Day","album":"Revolution Radio","duration":160,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:19YmvsVCetCBeVj6O2mljR","service":"spop","type":"song","name":"Still Breathing","title":"Still Breathing","artist":"Green Day","album":"Revolution Radio","duration":225,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:5JwuHzmJCmHuHGjP9W6KtK","service":"spop","type":"song","name":"Youngblood","title":"Youngblood","artist":"Green Day","album":"Revolution Radio","duration":153,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:3QWgLzCrAVkmw5eoHPN5uB","service":"spop","type":"song","name":"Too Dumb to Die","title":"Too Dumb to Die","artist":"Green Day","album":"Revolution Radio","duration":204,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:1bsQUnczB7JI4YnDzOjFMz","service":"spop","type":"song","name":"Troubled Times","title":"Troubled Times","artist":"Green Day","album":"Revolution Radio","duration":185,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:6wsfrOQOJK9CSwCsyMt1NJ","service":"spop","type":"song","name":"Forever Now","title":"Forever Now","artist":"Green Day","album":"Revolution Radio","duration":412,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"},{"uri":"spotify:track:35weSxL477aChs1wx7oZS5","service":"spop","type":"song","name":"Ordinary World","title":"Ordinary World","artist":"Green Day","album":"Revolution Radio","duration":183,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"}]
2016-10-28T20:11:32.882Z - info: CoreCommandRouter::volumioPushQueue
2016-10-28T20:11:32.885Z - info: [1477685492884] InterfaceWebUI::pushQueue
2016-10-28T20:11:32.962Z - info: [1477685492961] CoreStateMachine::updateTrackBlock
2016-10-28T20:11:32.964Z - info: [1477685492963] CorePlayQueue::getTrackBlock
2016-10-28T20:11:32.966Z - info: CoreCommandRouter::volumioPlay
2016-10-28T20:11:32.968Z - info: [1477685492968] CoreStateMachine::play
2016-10-28T20:11:32.970Z - info: [1477685492969] CoreStateMachine::setConsumeUpdateService undefined
2016-10-28T20:11:32.973Z - info: [1477685492972] CoreStateMachine::stop
2016-10-28T20:11:32.973Z - info: [1477685492973] CoreStateMachine::setConsumeUpdateService undefined
2016-10-28T20:11:32.975Z - info: [1477685492974] CoreStateMachine::play
2016-10-28T20:11:32.975Z - info: [1477685492975] CoreStateMachine::setConsumeUpdateService undefined
2016-10-28T20:11:32.976Z - info: [1477685492976] CorePlayQueue::getTrack 675
2016-10-28T20:11:32.978Z - info: [1477685492977] CoreStateMachine::startPlaybackTimer
2016-10-28T20:11:32.979Z - info: [1477685492978] CorePlayQueue::getTrack 675
2016-10-28T20:11:32.981Z - info: [1477685492981] ControllerSpop::clearAddPlayTrack
2016-10-28T20:11:32.982Z - info: {"uri":"spotify:track:4UGWMzkmEPpYoS9myk8lAG","service":"spop","type":"song","name":"Somewhere Now","title":"Somewhere Now","artist":"Green Day","album":"Revolution Radio","duration":249,"albumart":"/albumart?web=Green%20Day/Revolution%20Radio/large&path=","samplerate":"320Kbps","bitdepth":"16 bit","trackType":"spotify"}
2016-10-28T20:11:32.983Z - info: [1477685492982] ControllerSpop::sendSpopCommand
2016-10-28T20:11:32.984Z - info: ADDING DEFER FOR COMMAND uplay
2016-10-28T20:11:32.991Z - info: FIRST BRANCH
2016-10-28T20:11:32.992Z - info: BEFORE: SPOP HAS 1 PROMISE IN STACK
2016-10-28T20:11:32.992Z - info: AFTER: SPOP HAS 0 PROMISE IN STACK
2016-10-28T20:11:33.018Z - info: STATUS
2016-10-28T20:11:33.018Z - info: {
  "status" : "playing",
  "repeat" : false,
  "shuffle" : false,
  "total_tracks" : 1,
  "current_track" : 1,
  "artist" : "Green Day",
  "title" : "Somewhere Now",
  "album" : "Revolution Radio",
  "duration" : 249000,
  "position" : 0,
  "uri" : "spotify:track:4UGWMzkmEPpYoS9myk8lAG",
  "popularity" : 64
}

2016-10-28T20:11:33.019Z - info: 
[1477685493019] ---------------------------- Spop announces state update
2016-10-28T20:11:33.023Z - info: [1477685493023] ControllerSpop::parseState
[... cut ...]
2016-10-28T20:15:42.766Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2016-10-28T20:15:42.769Z - info: [1477685742767] InterfaceWebUI::pushState
2016-10-28T20:15:42.789Z - info: [1477685742787] ------------------------------ 111ms
2016-10-28T20:15:42.817Z - info: Pushing Favourites {"service":"spop","uri":"spotify:track:4UGWMzkmEPpYoS9myk8lAG","favourite":false}
2016-10-28T20:15:42.821Z - info: Pushing Favourites {"service":"spop","uri":"spotify:track:4UGWMzkmEPpYoS9myk8lAG","favourite":false}

fanciulli, 2016-11-03:
Hi,

in the log above i see

CorePlayQueue::getTrack 675

which tells me that Volumio is playing track number 675, probably the last one. If this is true there's no error in stopping playback after that song.

cortegedusage, 2016-11-03:
I got the same problem on all the 3 pi's with volumio 2.011 on it 
just playing one song of the playlist or album then nothing..

skikirkwood, 2016-11-03:
That's the strange thing. The Spotify playlist had 12 tracks. They were all added correctly and displayed in the queue. But Volumio always stops after playback of a single track.
  

  
I gave an extra Spotify Premium account you could use to test this but you'd probably need to VPN to a US ISP to use itvv

skikirkwood, 2016-11-03:
Here's a screenshot of the queue from the Spotify playlist added.  Volumio plays the first track and stops.
[image]

fanciulli, 2016-11-04:
Ok, i'll take a look at this issue asap

cortegedusage, 2016-11-04:
I don't now if it's related but it seems that playtime of the track is not always computed correctly
for instance if I take one track out of a list it displays time correctly at first, but when I skip to say 50 percent it looks like it calculates the time fault.
see printscreens underneath:
[image]

and at 50 percent:
[image]

I can repeat this randomly 

Like I said, don't now if it's related but it looks like it to me.

apart from that you're all doing a tremendous job, volumio2 has been 
a giant leap and still getting better, thanks

Dennis

cortegedusage, 2016-11-04:
when I play a song until it finishes, it doesn't seem to complete the last second, but stops anyway
see printscreen:
[image]

skikirkwood, 2016-11-08:
@fanciulli,I did a clean install of 2.001 on my RPi 3 and Spotify playlists play fine.  So this stopping after one track is a regression, which I believe showed up in 2.006, possibly earlier, but after 2.001.

fanciulli, 2016-11-08:
@skikirkwood the issue is due to a small change in the state machine we made. It has been fixed but it is currently in a development branch for analysis.

closed 2016-11-16 by volumio