volumio/Volumio2 #817: Disable SSH by default

bst27, 2016-10-31:
Volumio is advertised as _the simplest way to enjoy music_. Because of this I think it is a good idea to set end-user friendly defaults and ship the system in a secure state where this is possible without much effort. Therefore I suggest to disable SSH access by default.

An option to enable it could be added to the system settings in the WebUI or could be distributed as a plugin. I admit the fact that everyone is able to access the WebUI and enable SSH again is a problem I have no real answer for at the moment.

cgtobi, 2016-11-05:
If a proper password is set or even better preshared keys are in place it is secure enough and if the enduser does not want to use it, why care? SSH is not insecure because it's enabled, it's insecure because of the default password, just as in pretty much every other system.

Just my 2 cents.

bst27, 2016-11-06:
> SSH is not insecure because it's enabled, it's insecure because of the default password

I agree with you that SSH is not insecure per se. I just installed Volumio and discovered that it comes with an SSH server running and accepting credentials `volumio:volumio` by default.

I just wonder why the system is shipped to the end-user in such a vulnerable state. Maybe it would be better to hand out a more secure system by default and allow advanced users to enable SSH with secure credentials for expert customizations.

volumio, 2016-11-06:
SSH enabled is an important asset for both us developers and end users. The solution we will implement is a first-start wizard that will ask to change SSH password at first boot. I think this will be the best compromise beetween usability\security

closed 2016-11-06 by volumio