volumio/Volumio2 #819: Webradio track title not updated

cgtobi, 2016-10-31:
I'm currently running v2.011. I compared the same stream (http://streams.radiobob.de/bob-alternat ... ediaplayer) in Theremin (macOS mpd client), VLC, MPoD (iOS mpd client) and Volumio. MPoD, VLC and Theremin update the currently playing title fine and VLC (macOS) as well as MPoD v2.1 (iOS) show the stream title, whereas Volumio stops updating the currently playing title and doesn't display the stream title at all.

[image]

daeavelwyn, 2016-11-18:
hi here,

as mentionned in the forum ([https://volumio.org/forum/001-webradio-track-name-not-updated-t4925.html](https://volumio.org/forum/001-webradio-track-name-not-updated-t4925.html), this bug is still present in the 2.029 update :)

Loader23, 2016-11-25:
Also present in 2.031.

cgtobi, 2016-11-28:
As far as I can tell volumio knows all about what's beeing played, so it must be some communication or display issue in the front end that the tracks are not updated correctly.

e.g.
`Nov 28 21:24:52 volumio volumio[816]: info: STATE SERVICE {"status":"play","position":0,"seek":2444678,"duration":0,"samplerate":"44.1 KHz","bitdepth":"24 bit","channels":2,"random":false,"updatedb":false,"repeat":false,"isStreaming":false,"title":"Faith No More - I'm Easy","artist":null,"album":null,"uri":"http://streams.radiobob.de/bob-kuschelrock/mp3-192/mediaplayer","trackType":"de/bob-kuschelrock/mp3-192/mediaplayer"}
`

Loader23, 2016-12-18:
Still present in 2.041...

biva, 2017-03-15:
Still present in 2.118. Any update? Available for testing if needed.

xipmix, 2017-04-06:
There is one thing that doesn't look right to me:
 *  ```"trackType":"de/bob-kuschelrock/mp3-192/mediaplayer"}```
This seems to be the same symptom that I noticed in #1116. 

A similar issue was fixed between 2.099 and 2.114 (somewhere between 29 Feb and 6 Mar) for the 'Artist' field, but here we have ```"artist":null```. This is probably a property of the webradio source so volumio is probably doing the right thing for that field.

volumio, 2017-06-13:
It now works

closed 2017-06-13 by volumio

jlagden, 2018-04-28:
I am finding the title is not getting updated when I play internet radio streams.
eg. http://138.201.248.219:8018/stream

If I play in winamp on a pc desktop the title gets updated, however it doesn't in volumio.
I noticed if I do a `mpc status` mpd is not updating the title either, so this is probably a mpd problem.

I am using Volumio 2.389 (latest)

Any thoughts?