volumio/Volumio2 #822: Loosing network connectivity after some time

heis2201, 2016-11-02:
Generally my volumio2 is working fine on my Rpi B+ with Hifiberry DAC (actually Durio Sound), but after some uptime I loose network connection (connected wirelessly). I can't neither ssh not http into volumio. Sadly, all that helps is a hard poweroff. Volumio keeps playing my playlist, but won't reply any request at all.

What can be the reason here, and how can I debug it when I am not able to ssh into it?

exam22, 2016-11-26:
Same with me. I use ethernet but it doesn't appear on my router.
Even I use jump wire to my main PC, not got response everytime.

heis2201, 2016-11-26:
It happens to me as we speak. Just wanted to change playlist, but Volumio is not reachable.
I wish I knew how to debug or fix this problem...

exam22, 2016-11-26:
I've change to static ip then it become much stable.
But before I can access setting with WebUI, I've reboot lots of times....

closed 2017-06-03 by volumio