volumio/Volumio2 #824: Can't reach webserver after update

Bonnee, 2016-11-02:
After I updated to RC2 through the WebUI I encountered this error in the systemd status output for volumio.service:

> volumio[2843]: Discovery: StartAdv! false
> volumio[2843]: Discovery: Started advertising... Volumio - false
> volumio[2843]: Discovery: ecc TypeError: Cannot read property '0' of undefined

This error comes up repeatedly when I try to load the WebUI inside a browser.
Is there a way to fix this? I'll post more if someone instructs me on what to do

Thanks!

volumio, 2016-11-02:
This is a dns related error. Remove power and turn it on again, it should disappear

Bonnee, 2016-11-02:
Unfortunately I rebooted many times and the error didn't disappear. I also noticed that my / and /static partitions are full. Can that be the cause?

heis2201, 2016-11-06:
I'd recommend you start with a newer release than RC2. For example, 2.001: https://volumio.org/get-started

Bonnee, 2016-11-06:
That's just what I did yesterday and now it works!
Thanks for the help!

closed 2016-11-06 by Bonnee