volumio/Volumio2 #830: Shuffle does not function properly using volumio UI

Shodan0000, 2016-11-05:
When multiple albums are added to a playlist and shuffle is activated, it only shuffles the first album. Only when manually a track is skipped songs from other albums get played.

youpilai, 2016-11-05:
Can't tell for the album part, but when using personal playlist, it only random when pressing the "next track" button.
When finishing a track, it plays the next in playlist, no random there

arthurlutz, 2016-11-07:
same here, I think @youpilai 's description is what I am experiencing.

Bonnee, 2016-11-17:
I've got the same issue as @youpilai on a freshly installed 2.001 on a RPi B

NeoArmageddon, 2016-11-19:
Same here. Only "Next"-button seem to select a random song from the playlist. When left alone, the playlist is played in order.

chiu0602, 2018-03-11:
My workaround is, putting a blank mp3 file be the first file to play
[https://github.com/anars/blank-audio](https://github.com/anars/blank-audio)

xipmix, 2018-04-12:
I tried to reproduce this on 2.389 and had trouble doing so. 
Is it still an issue? The state machine got a tuning up a couple of months ago so it may have improved.

If you anyone is hitting this, please post [logs](https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html) and give some indication of how you tested, so someone can try to reproduce the issue.

How I tested:
Test 1:
 * clear queue
 * add 5 tracks one at a time, from one album to the queue
 * press 'random' on the queue page
 * press 'play' on the queue page
Result: Tracks 2,4,2,3,4,5 were played, then the playback stopped. It played track 2 twice and did not play track 1, Which is odd but different to the sequential behaviour described here.

Test 2:
* clear queue
* add 3 tracks, one at a time, from one album to the queue
* add 3 tracks, one at a time, from a different album (different artist, too), to the queue
* press 'random' on the queue page
* press 'play' on the queue page
Result: Tracks 5,6 were played, then the player stops. This is more like what has been described.

Test 3:
* After test 2 completed, press Play again, on the queue page.
Result: Tracks 1,5,2,4,3,6 played. Which could be random or could be an interleaving of 1,2,3 and 4,5,6

I took care to not press any buttons like 'next' etc during these, to avoid introduce other code paths.

@chiu0602 - can you send logs with and without the blank mp3 file? That would be rather interesting.