volumio/Volumio2 #833: update to 2.013 brakes system

cortegedusage, 2016-11-05:
was trying to update my system to 2.013 but it didn't came back after reboot
i could ssh into the system but web service didn't came up.
reflashed the card and tried the same thing again, with the same result.

sticking to  2.001 for now,

using pi3 with i2s dac

cortegedusage, 2016-11-06:
noticed a new update to 2.014 dit an update from 2.001 
same result, earlier I could update without a problem.

Had only spotify connect installed(latest version)
although yesterday I tried to update direct after a fresh install, same result...

any light on this?

volumio, 2016-11-06:
You updated to a test build, and we told many times that this is not encouraged because it might break your system. I'm rebuilding a new image now which should work.

closed 2016-11-06 by volumio

cortegedusage, 2016-11-06:
I know it's a test build, and I wasn't posting because I'm angry or impatience.
I was just posting it because I thought it was a bug of which I should inform you
to improve the software.

maybe I shouldn't have post it here on github but on the volumio forum.

Anyways, keep up the good work, it's a good thing you have here..

dennis

volumio, 2016-11-06:
Thank you Dennis, I must have bad-interpreted your comment. I'm testing a new build as we speak..

cortegedusage, 2016-11-06:
no problem, short messages like these are often bad interpreted because
there is simply no context at all.

keep up the good work.

Dennis