volumio/Volumio2 #849: Web Interface has many bugs, trying to control Spotify from its native app instead?

misaghshakeri, 2016-11-15:
The web app has many issues. specially cache problems. my long playlists cannot even be shown.
I was thinking about listing these issues and trying to resolve them but there is a better solution:
Spotify normally detects devices to which it is broadcasting. so you can control the streaming from the native Spotify app on your phone or on your desktop. 
But bizarrely it doesn't detect Volumio. does anyone have any idea what is wrong with it ?

skikirkwood, 2016-11-18:
You are referring to the Spotify Connect protocol.  A plugin that supports Spotify Connect is currently under development:

https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect

misaghshakeri, 2016-11-19:
@skikirkwood yes thanks ! I would like to help :)

closed 2016-11-19 by misaghshakeri