volumio/Volumio2 #856: NAS size error: NaN when path has spaces

macmpi, 2016-11-18:
There seem to be a nasty bug in NAS shares size extraction [in here](https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/networkfs/index.js#L523).
Indeed when share path has spaces in it, then `awk` got mixed-up, and does not returns the proper column!...
Following size calculation is then altered and may end-up with NaN, which is what got displayed in UI.

xipmix, 2016-11-19:
This should fix the issue with spaces in the mountpoint
```
- var cmd="df -BM "+mountpoint+" | awk '{print $3}'";
+ var cmd="df -BM --output=used '"+mountpoint+"' | tail -1 | awk '{print $1}'";
```
but ... does the code (in current git) that sets up shares allow spaces in the name?

macmpi, 2016-11-19:
Thanks a lot: will try that.
I was not referring to spaces in the `mountpoint` name per say, but spaces within the mounted server path (like `192.168.3.200:Multimedia/MUSIC/iTunes Master/iTunes Media/Music` fairly common in iTunes for instance).
Paths with spaces seem supported (changelog in recent 2.0.29 `Fix share with spaces` makes it even more explicit).

volumio, 2016-11-19:
Good tip! I'll try it now

2lepus, 2016-11-19:
When the mounted folder has no spaces but the subfolders have, then you get 0.00MB instead

2lepus, 2016-11-19:
Ok, though I deleted that mount and mounted it again I still get 0.00MB or something smaller than the real size.
(with 2.0.29)
[image]

xipmix, 2016-11-20:
> I was not referring to spaces in the mountpoint name per say, but spaces within the mounted server path (like 192.168.3.200:Multimedia/MUSIC/iTunes Master/iTunes Media/Music fairly common in iTunes for instance).

Ah, yes. I was worried about that too. Using ```--output=used``` works around that by not printing the paths at all.

macmpi, 2017-02-17:
With https://github.com/volumio/Volumio2/pull/879 closed, is this original issue meant to be fixed as a consequence of https://github.com/volumio/Volumio2/pull/1010 commit, or is this issue is still waiting for pending resolution (and updated PR)?
Thanks to close the issue if now fixed.

xipmix, 2017-02-17:
I plan to submit a new PR that improves the df call but drops the rest of it, I'm guessing there was some discomfort with the number formatting stuff.

PR #1017 (2017-02-20): Make retrieval of share size a little more robust.
Quote the NAS mountpoint when calling 'df' and return just the wanted value
with no units, so computing a displayed size becomes simpler.

Fixes: #856
closed 2017-02-20 by volumio