volumio/Volumio2 #860: Back button in BROWSE environment for Spotify What's new goes to far back

WalterBorgstein, 2016-11-20:
While searching for music in the browse section, in Spotify, What's. You can follow a path down to a specific number of an album (see screenshots below). If however you decide you want one step back. back to the album overview, the back button will start at the beginnen of the browser view. This makes browsing thru new music in Spotify not user friendly.

This only seems to happen in the What's new section of Spotify.

I saw not difference using Google Element inspector. But I'm not a dev.

[image]
[image]
[image]
[image]

xipmix, 2017-06-02:
possibly related issues: #498, #975

volumio, 2017-06-13:
Duplicate, closing

closed 2017-06-13 by volumio