volumio/Volumio2 #863: Bad documentation

misaghshakeri, 2016-11-20:
it's been two days that I am trying to just mount the volumio virtual machine and figure out how it works. is it me or the documentation is not clear ? how can I just set the development environment ? it should be simple but it is not. how can we improve that ?

volumio, 2016-11-20:
Actually, it is suggested to develop on the pi itself. We made a small guide here:
https://github.com/volumio/Volumio2/wiki/Setting-Up-a-Dev-Environment

But I'll take the committment to add everything to the doc. Just let me know what you need.

In any way, better to develop on the pi or on the real device you're using... Let me know

misaghshakeri, 2016-11-21:
Thanks, 
I will setup the dev environment and I let you know :) 
My first impression is that we can improve the documentation. but I come back to it once I set my dev environment.

skikirkwood, 2016-12-03:
I made a short video that shows how I set up debugging of Volumio 2 on a Pi 3 from my Mac using PHPStorm: https://volumio.org/forum/debugging-volumio-node-code-with-webstorm-t4347.html

jotak, 2016-12-20:
I suggest large cuts in the readme (because it overlaps with the documentation website), and also removing old TODO list that doesn't seem relevant anymore. Check on my fork: https://github.com/jotak/Volumio2 ; what do you think?

volumio, 2016-12-20:
Absolutely agree! Could you send a PR for the readme?

misaghshakeri, 2016-12-20:
@jotak great job ! it is much more readable now.

closed 2017-06-02 by volumio