volumio/Volumio2 #864: Focusing on the backend

misaghshakeri, 2016-11-20:
Guys, I am trying to contribute to this project, as I really love it. but I have an existential question: 

From where I sit, Volumio is a fabulous backend project, personally I have never seen such a high quality backend media center. it can beat most of the commercial solutions in the market.  why should we bother developing a front end for it ? WebUI ?! 
it is not the same project. the community formed here is back-end oriented and they are good at it. I feel that we are trying to develop and debug a WebUI which has bugs and hardly useable since at least 3 years ago (I follow it since then).

Why can't we focus just on back-end ? we can play songs on the command line right ? who cares about the artwork ? 

at least we can separate these two parts. one project: back-end API based player, and the other part WebUI.

Don't tell me that it is already separated, cause it is not. I expect a backend independent Volumio with a documentation like: 

Log into the system and type: 

> volumio play -spotify -songID="https://open.spotify.com/track/6ZrauffaDWD1kaN4Z6pFVh" 
Then what happens is: 

1) Spotify plugin fetch the ID 
2) SPOP plays .. 

And so on ... 

There are still so many features that can be added to backend, and we are struggling with the front end, which is not an interesting project from many people in this community. 

in my humble opinion we should build a community dedicated to backend.

volumio, 2016-11-20:
Thank you so much for your  appreciation! 
To reply to your points:
- The UI and the backend (in Volumio2) are meant to be two completely different and detached systems. The fact that they are developed together is because we think that it will be the way most people use volumio. But Volumio can and will work even withuout a UI, which is just one of the possible user interfaces
- The "user interface" type plugins are meant to control the backend. Now we do have the websocket API, the REST API and a command line client (which uses the websocket api). We can add as many user interfaces as we want (like gpios, keyoboard, lirc etc). 
- If you're interested to create a commandline client, we already started and you can further develop it.
See here:
https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/volumio_command_line_client/volumio.sh

You can see the existing APIs here:
https://volumio.github.io/docs/API/WebSocket_APIs.html
https://volumio.github.io/docs/API/REST_API.html

let me know what you think!

misaghshakeri, 2016-11-21:
Thank you, 
Yes I see your logic, and I think that you are right. the problem is just the way the repository and wikis are organized. 
For instance, Volumio2 (this project) is supposed to be the backend. but in its wiki we are talking about Angular and Express which is a little confusing. 
I think that it would be more efficient and understandable if each repository focus only on one thing having a clear objective: 

- Volumio OS
- Core
- WebUI

If we want to have a global wiki explaining the overall architecture it can be a link at the end of each project wiki pointing to a document on volumio blog.

closed 2017-06-02 by volumio