volumio/Volumio2 #874: alsamixer Master set to 0 after clean install

lsinger, 2016-11-23:
Version: 2.031
Pi3 and Hifiberry Amp+

1. downloaded image and wrote to SD card
2. after first boot used hotspot to enter my WiFi settings
3. rebooted
4. `Playback Options / Output Device` is set to Audiojack; Amp is not auto-detected
5. turned on I2S DAC option in `Playback Options / Output Device` and manually selected Amp
6. rebooted; startup sound was audible loud and clear
7. tried to play soething via Airplay, no success

After this I tried fiddling around with settings a lot until I finally thought of SSH'ing into it and checking `alsamixer`. And indeed, it turns out the Amp's Master volume is set to 0. 

Expected: the Amp's Master volume to be set to a reasonable level

Do let me know if there's any other way for me to help get to the bottom of this. Thanks for all your hard work on volumio. ❤️

Jan-Bart, 2016-12-09:
Version: 2.031
Pi2B and Hifiberry Amp+

I'm having problems too after a clean install and it looks related.
In the Volumio interface I can change volume between 0 and 10. (but not above 10)

If I use alsamixer (as stated by @lsinger) I have full control of volume.

closed 2017-06-02 by volumio

lsinger, 2017-06-03:
@volumio I assume you can't reproduce?