volumio/Volumio2 #878: Mount promblem after update to Volumio2

exam22, 2016-11-26:
When I mount in My Music, it shows error code below:
Error in mounting share yourname
(112): Host is down Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) 
or
Error in mounting share Music_On_PC
(11): Resource temporarily unavailable Refer to the mount.cifs(8) manual page (e.g. man mount.cifs) 
It works fine in Volumio with Cubietruck but always error in Volumio2 (2.03).
Should I change any setting in win10？

codewarriorFX, 2016-12-05:
Im having this same error on my Odroid C2 (Volumio 2.03) and my NAS is Windows 10.

Where you able to fix this issue?

xipmix, 2016-12-06:
Firstly, is the error shown in the web UI or in the system log?

If the latter, can you log in and collect the output from ```sudo journactl``` around the time the mount tries and fails? This will show the exact mount command that was used.

Are you using an IP address or hostname for the NAS in the 'IP address field' - it may not be able to resolve the name.

exam22, 2016-12-06:
https://drive.google.com/file/d/0B3jXBnQTA1ROTW1naGZzdzMzeU0/view?usp=sharing
I've copy it all to txt file.

xipmix, 2016-12-07:
I think the main problem is that you have spaces in the path to the share; the version of volumio you are on does not handle this. I don't think those fixes have made it in yet.

xipmix, 2016-12-07:
After looking more carefully, it might be using a hostname instead of an IP address, though the error messages are misleading. Here's the whole sequence collated for a share with a hostname
```
Nov 25 11:58:39 volumio sudo[913]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/mount -t cifs -o guest,ro,dir_mode=0777,file_mode=0666,           iocharset=utf8,noauto //DESKTOP-BNFKF26/Music_On_PC /mnt/NAS/Music_On_PC
Nov 25 11:58:39 volumio sudo[913]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 25 11:58:39 volumio sudo[913]: pam_unix(sudo:session): session closed for user root
Nov 25 11:58:40 volumio volumio[885]: exec error Error: Command failed: /usr/bin/sudo /bin/mount -t cifs -o guest,ro,dir_mode=0777,file_mode=0666,            iocharset=utf8,noauto //DESKTOP-BNFKF26/Music_On_PC /mnt/NAS/Music_On_PC
Nov 25 11:58:40 volumio volumio[885]: mount error(13): Permission denied        
Nov 25 11:58:40 volumio volumio[885]: Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
Nov 25 11:58:40 volumio volumio[885]: info: Cannot mount NAS Music_On_PC at system boot, trial number 1 ,retrying in 5 seconds
```

Here's the same share with an IP address, there's no corresponding 'permission denied' in the logs
```
Nov 25 11:58:39 volumio sudo[922]: volumio : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/mount -t cifs -o username=GUEST,password=,ro,dir_mode=0777,       file_mode=0666,iocharset=utf8,noauto //192.168.1.100/Music_On_PC /mnt/NAS/Music_On_PC
Nov 25 11:58:39 volumio sudo[922]: pam_unix(sudo:session): session opened for user root by (uid=0)
Nov 25 11:58:39 volumio sudo[922]: pam_unix(sudo:session): session closed for user root
```
There are definitely some problems when adding with spaces in the share name, fixes for that should not be far away.

xipmix, 2016-12-07:
So by now you are wondering why using a hostname, which probably used to work, does not work.
There have been some changes to the way volumio does dhcp and the default settings in /etc/resolv.conf that are likely to be the cause. Can you post what you have in there?
```
volumio$ cat /etc/resolv.conf
```
and whether the name resolves
```
volumio$ host DESKTOP-BNFKF26
```

exam22, 2016-12-07:
/etc/resolv.conf 
# Generated by dhcpcd from eth0
# Google nameservers
nameserver 8.8.8.8
nameserver 8.8.4.4
nameserver 0.0.0.0
nameserver 8.8.8.8
# /etc/resolv.conf.tail can replace this line

Actually, I'm using static IP in webui setting because this issue:
https://github.com/volumio/Volumio2/issues/822#issuecomment-263043150
I'll make a clean Volumio if I'm lucky to get in WebUI.

I've find I can see NAS when I don't plug USB Drive in.
https://drive.google.com/file/d/0B3jXBnQTA1ROQXhpRWMzVlpsYkU/view?usp=sharing
But only "Music_On_PC" has files.
After I click play, it start five second and become silence, then I can't find NAS anymore,
until next time Rescan.

volumio, 2016-12-07:
it seems that your mpd is crashing

xipmix, 2016-12-07:
> it seems that your mpd is crashing

There are two other odroid users reporting this also, #777, #890.
No idea what the real cause is.

--

volumio, 2016-12-07:
I am trying to find it out as well, recompiling mpd to see what it is

exam22, 2016-12-08:
I've use clean Volumio to mount no spaces files.
They shown in NAS folder, but when open each folde, no files in them.
This is log, Thanks.
https://drive.google.com/file/d/0B3jXBnQTA1ROd0Y5SkpFQ0toQUU/view?usp=sharing

bdouble, 2016-12-09:
@volumio this is happening on cubox-i too, #890 is a report on an on that platform (a cubox-i4 in my case).

exam22, 2016-12-30:
Hello.
I think I found the problem may be my Antivirus software(Bitdefender and Avast)
After remove them, I can mount files successful.
Seems mount file can't over 5000 files, could it get improve?
Thanks a lot.

exam22, 2017-01-02:
https://ask.fedoraproject.org/en/question/87463/samba-windows-10-share-mount-error112-host-is-down/
I think maybe it could try.

closed 2017-06-02 by volumio

gerritv, 2017-08-05:
I can verify that the solution in the link from @exam22 works. No one should be using SMB1 anymore after the debacle of WannaCry etc. Please release a version of Volumio with ver-2.0 or ver=3.0 option added to the mount. I am using Volumio v2.224 and it is not in that version so not sure why this Issue was Closed

A workaround is to add ver=2.0 to Options in Advanced options section of Add New Drive.

gerritv, 2017-08-05:
I can verify that the solution in the link from @exam22 works. No one should be using SMB1 anymore after the debacle of WannaCry etc. 
Please release a version of Volumio with ver-2.0 or ver=3.0 option added to the mount. I am using Volumio v2.224 and it is not fixed in that version so not sure why this Issue was Closed

A workaround is to add ver=2.0 to Options in Advanced options section of Add New Drive.
--

volumio, 2017-08-05:
You can add it manually via advanced options. We need to care about compatibility with everyone, even if they use stone age stuff

gerritv, 2017-08-05:
Fair enough. I will add it to the CIFS How To on the forum to make it more visible.