volumio/Volumio2 #890: System Crash During Library Scanning on Cubox-i

bdouble, 2016-11-29:
Volumio 2.029 cannot complete a scan of a large >1.5 TB library on a Synology NAS. I can mount the drive successfully and begin a scan, but sometime around 60-100 albums in and the system locks up. It won't respond to any network requests, and the shell becomes unresponsive. All volumio services seem to halt. It does not appear to be related to any specific files as the kernel error happens scanning different albums and files every time. 

Sample output of journalctl -f:
Nov 29 05:12:58 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/Sacrebleu/15 - Encore un Terlude.mp3
Nov 29 05:12:58 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/Sacrebleu/08 - Dirty Larry (edit).mp3
Nov 29 05:12:59 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/10 - Love Boat Theme.mp3
Nov 29 05:12:59 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/01 - Seconds (feat. Loleatta Holloway) (12_ version).mp3
Nov 29 05:12:59 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/12 - Catch Me on the Rebound (Walter Gibbons mix).mp3
Nov 29 05:12:59 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/03 - Love Sensation (Shep Pettibone Rough mix under Dim's DJ Friendly re-edit).mp3
Nov 29 05:13:00 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/15 - Let No Man Put Asunder.mp3
Nov 29 05:13:00 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/02 - Moment of My Life (feat. Jocelyn Brown).mp3
Nov 29 05:13:00 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/04 - (Knock Out) Let's Go Another Round (feat. Jocelyn Brown).mp3
Nov 29 05:13:00 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/05 - Here's to You (Dim's Body & Soul re-edit).mp3
Nov 29 05:13:00 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/07 - Sadie (She Smokes).mp3
Nov 29 05:13:00 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/08 - Ohh, I Love It (Love Break) (Dimitri re-edit).mp3
Nov 29 05:13:00 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/14 - Ten Percent.mp3
Nov 29 05:13:00 volumio kernel: Alignment trap: not handling instruction e191cf9f at [<c01fb24c>]
Nov 29 05:13:00 volumio kernel: Unhandled fault: alignment exception (0x001) at 0x55cb8b9a
Nov 29 05:13:00 volumio kernel: pgd = ed334000
Nov 29 05:13:00 volumio kernel: [55cb8b9a] *pgd=00000000
Nov 29 05:13:00 volumio kernel: Internal error: : 1 [#1] SMP ARM
Nov 29 05:13:00 volumio mpd[620]: update: added NAS/Vault/Dimitri From Paris/My Salsoul_ Selected and Mixed by Dimitri From Paris/06 - Boogie's Gonna Get Ya (instrumental_dub).mp3

bdouble, 2016-12-02:
getting a different sort of error by running mpc update on the command line:

Dec 02 04:45:18 volumio mpd[611]: update: added NAS/vault/Freddie Mercury/Mr. Bad Guy/03 - I Was Born to Love You.mp3
Dec 02 04:45:19 volumio mpd[611]: update: added NAS/vault/Freddie Mercury/Mr. Bad Guy/08 - There Must Be More to Life Than This.mp3
Dec 02 04:45:19 volumio mpd[611]: update: added NAS/vault/Freddie Mercury/Mr. Bad Guy/01 - Let's Turn It On.mp3
Dec 02 04:45:19 volumio mpd[611]: update: added NAS/vault/Freddie Mercury/Mr. Bad Guy/02 - Made in Heaven.mp3
Dec 02 04:45:19 volumio mpd[611]: update: added NAS/vault/Grateful Dead/Shakedown Street/15 - All New Minglewood Blues (live).mp3
Dec 02 04:45:20 volumio mpd[611]: update: added NAS/vault/Grateful Dead/Shakedown Street/14 - Stagger Lee (live).mp3
Dec 02 04:45:20 volumio mpd[611]: update: added NAS/vault/Grateful Dead/Shakedown Street/08 - Stagger Lee.mp3
Dec 02 04:45:20 volumio mpd[611]: update: added NAS/vault/Grateful Dead/Shakedown Street/13 - Fire on the Mountain (live).mp3
Dec 02 04:45:20 volumio mpd[611]: update: added NAS/vault/Grateful Dead/Shakedown Street/05 - Fire on the Mountain.mp3
Dec 02 04:45:21 volumio kernel: Unable to handle kernel paging request at virtual address 58ea82be
Dec 02 04:45:21 volumio kernel: pgd = ebed4000
Dec 02 04:45:21 volumio kernel: [58ea82be] *pgd=00000000
Dec 02 04:45:21 volumio kernel: Internal error: Oops: 5 [#1] SMP ARM
Dec 02 04:45:21 volumio kernel: Modules linked in: nfsd nfs_acl xt_REDIRECT nf_nat_redirect iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack ip_tables brcmfmac brcmutil dw_hdmi_ahb_audio evbug squashfs overlay

xipmix, 2016-12-04:
At first glance this appears to be a duplicate of #777. But with the mpc client you are seeing a kernel oops which is more serious. So there are probably bugs in both mpd (incorrectly aligned memory access) and the kernel (unable to access memory it should have been able to). This could be a hardware issue, or a kernel bug.

[This commit](http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=20d7a35bc8be06dd1350de0454b0e24cca1d4354) (one of a series starting [here](http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=cf61e2a1487d833e4748dead4096584de70bf742)) might be addressing the problem you are seeing. Looks like it would have gone into kernel 4.6, which I think is later than the one in volumio.

To move forward from here probably requires some work with gdb and/or strace. It would be interesting to know if @RastaX sees 'mpc update' fail in the same way you do.

RastaX, 2016-12-04:
This is the error and output i get, when i start mpc update on the command line - the error comes with a complete restart of UI:

Dec 04 09:53:29 volumio kernel: Unhandled fault: alignment fault (0x92000021) at 0x00000000eeb1c107
Dec 04 09:53:29 volumio systemd[1]: mpd.service: main process exited, code=killed, status=7/BUS
Dec 04 09:53:29 volumio systemd[1]: Unit mpd.service entered failed state.
Dec 04 09:53:29 volumio systemd[1]: mpd.service holdoff time over, scheduling restart.
Dec 04 09:53:29 volumio systemd[1]: Stopping Music Player Daemon...
Dec 04 09:53:29 volumio systemd[1]: Starting Music Player Daemon...
Dec 04 09:53:29 volumio systemd[1]: Started Music Player Daemon.
Dec 04 09:53:29 volumio mpd[1068]: zeroconf: No global port, disabling zeroconf
Dec 04 09:53:30 volumio volumio[896]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Dec 04 09:53:30 volumio volumio[896]: MPD error: Error: This socket has been ended by the other party
Dec 04 09:53:30 volumio volumio[896]: MPD error: Error: This socket has been ended by the other party
Dec 04 09:53:30 volumio volumio[896]: MPD error: Error: This socket has been ended by the other party
Dec 04 09:53:34 volumio volumio[896]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Dec 04 09:53:34 volumio volumio[896]: MPD error: Error: This socket has been ended by the other party
Dec 04 09:53:34 volumio volumio[896]: MPD error: Error: This socket has been ended by the other party
Dec 04 09:53:34 volumio volumio[896]: MPD error: Error: This socket has been ended by the other party
Dec 04 09:53:38 volumio volumio[896]: info: CoreCommandRouter::executeOnPlugin: mpd , getMyCollectionStats
Dec 04 09:53:38 volumio volumio[896]: assert.js:85
Dec 04 09:53:38 volumio volumio[896]: throw new assert.AssertionError({
Dec 04 09:53:38 volumio volumio[896]: ^
Dec 04 09:53:38 volumio volumio[896]: AssertionError: false == true
Dec 04 09:53:38 volumio volumio[896]: at MpdClient.sendCommand (/volumio/node_modules/mpd/index.js:86:10)
Dec 04 09:53:38 volumio volumio[896]: at ControllerMpd.getMyCollectionStats (/volumio/app/plugins/music_service/mpd/index.js:2256:20)
Dec 04 09:53:38 volumio volumio[896]: at CoreCommandRouter.executeOnPlugin (/volumio/app/index.js:818:29)
Dec 04 09:53:38 volumio volumio[896]: at Socket.<anonymous> (/volumio/app/plugins/user_interface/websocket/index.js:1112:43)
Dec 04 09:53:38 volumio volumio[896]: at emitTwo (events.js:106:13)
Dec 04 09:53:38 volumio volumio[896]: at Socket.emit (events.js:191:7)
Dec 04 09:53:38 volumio volumio[896]: at Socket.onevent (/volumio/node_modules/socket.io/lib/socket.js:348:8)
Dec 04 09:53:38 volumio volumio[896]: at Socket.onpacket (/volumio/node_modules/socket.io/lib/socket.js:308:12)
Dec 04 09:53:38 volumio volumio[896]: at /volumio/node_modules/socket.io/lib/client.js:195:16
Dec 04 09:53:38 volumio volumio[896]: at _combinedTickCallback (internal/process/next_tick.js:67:7)
Dec 04 09:53:38 volumio systemd[1]: volumio.service: main process exited, code=exited, status=1/FAILURE
Dec 04 09:53:38 volumio systemd[1]: Unit volumio.service entered failed state.
Dec 04 09:53:38 volumio systemd[1]: Starting dynamicswap.service...
Dec 04 09:53:38 volumio systemd[1]: Started dynamicswap.service.
Dec 04 09:53:39 volumio systemd[1]: volumio.service holdoff time over, scheduling restart.
Dec 04 09:53:39 volumio systemd[1]: Starting dynamicswap.service...
Dec 04 09:53:39 volumio systemd[1]: Started dynamicswap.service.
Dec 04 09:53:39 volumio systemd[1]: Stopping Volumio Backend Module...
Dec 04 09:53:39 volumio systemd[1]: Starting Volumio Backend Module...
Dec 04 09:53:39 volumio systemd[1]: Started Volumio Backend Module.
Dec 04 09:53:41 volumio volumio[1087]: info: -------------------------------------------
Dec 04 09:53:41 volumio volumio[1087]: info: -----            Volumio2              ----
Dec 04 09:53:41 volumio volumio[1087]: info: -------------------------------------------
Dec 04 09:53:41 volumio volumio[1087]: info: -----          System startup          ----
Dec 04 09:53:41 volumio volumio[1087]: info: -------------------------------------------
Dec 04 09:53:41 volumio volumio[1087]: info: Plugin folders cleanup
Dec 04 09:53:41 volumio volumio[1087]: info: Scanning into folder /volumio/app/plugins/
Dec 04 09:53:41 volumio volumio[1087]: info: Scanning category audio_interface

RastaX, 2016-12-04:
I'm back on version "VolumioRC2Fix-2016-07-31-odroidc2" again, which scans through the directory without problems!

I could post kernel versions of my RC2-system, if that helps and if you guide me with the necessary commands...

closed 2018-03-26 by volumio