volumio/Volumio2 #895: Edimax 802.11ac USB dongle does not work with Volumio 2 (EW-7811utc)

chrismhinds, 2016-11-30:
Hi - I've been using Volumio 1 for at least a year now, and I could use my Edimax 802.11ac USB dongle with Volumio.  Now, in Version 2, I cannot.

When I initially attempted to install Volumio 1, my USB dongle didn't work.  However, I went into IRC, and within a couple hours, we had found a workaround ... and, in subsequent versions, support for this USB wifi dongle was included in updates. So, I made a mistake and blindly installed V2 over V1 without backing it up.  Oops!

Here are some relevant bits:
volumio@volumio:~$ lsusb
Bus 001 Device 005: ID 7392:a812 Edimax Technology Co., Ltd
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

volumio@volumio:~$ cat /etc/network/interfaces
auto wlan0
auto lo
iface lo inet loopback

allow-hotplug eth0
iface eth0 inet dhcp

allow-hotplug wlan0
iface wlan0 inet manual

----
Related to the last fix was something along these lines:
[ https://ubuntuforums.org/showthread.php?t=2228244&p=13042921#post13042921 ]

    sudo apt-get update
    sudo apt-get install linux-headers-generic build-essential git
    git clone https://github.com/gnab/rtl8812au.git
    cd ~/rtl8812au
    make
    sudo make install
    sudo modprobe 8812au

I can update, install git, and clone the respository.  However, when I attempt to make, I get this:

    volumio@volumio:~/rtl8812au$ sudo ./install.sh
    make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.4.9-v7+/build M=/home/volumio/rtl8812au  modules
    make[1]: *** /lib/modules/4.4.9-v7+/build: No such file or directory.  Stop.
    Makefile:1052: recipe for target 'modules' failed
    make: *** [modules] Error 2

Thoughts/suggestions?  Thanks!

volumio, 2016-11-30:
launch this before compiling

volumio kernelsource

let me know

chrismhinds, 2016-11-30:
Aha!  We're making progress.  Thank you.  Incidentally, I found another location for the .ko file, and I used this link:

    wget https://dl.dropboxusercontent.com/u/80256631/8812au-4.4.9-v7-884.tar.gz

And now, for my next trick, I will figure out why the wireless adapter is attempting to use a different subnet:

    volumio@volumio:~$ ifconfig
    eth0      Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx
              inet addr:192.168.1.193  Bcast:192.168.1.255  Mask:255.255.255.0
              UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
              RX packets:444 errors:0 dropped:1 overruns:0 frame:0
              TX packets:262 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000
              RX bytes:110496 (107.9 KiB)  TX bytes:43774 (42.7 KiB)
    lo        Link encap:Local Loopback
              inet addr:127.0.0.1  Mask:255.0.0.0
              UP LOOPBACK RUNNING  MTU:65536  Metric:1
              RX packets:465 errors:0 dropped:0 overruns:0 frame:0
              TX packets:465 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1
              RX bytes:44378 (43.3 KiB)  TX bytes:44378 (43.3 KiB)
    wlan0     Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx
              inet addr:192.168.211.1  Bcast:192.168.211.255  Mask:255.255.255.0
              UP BROADCAST MULTICAST  MTU:1500  Metric:1
              RX packets:0 errors:0 dropped:6 overruns:0 frame:0
              TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
              collisions:0 txqueuelen:1000
              RX bytes:532 (532.0 B)  TX bytes:0 (0.0 B)

chrismhinds, 2016-11-30:
After monkeying around with it for a few minutes, I just assigned it a static IP using the volumio ->network page, and I'm calling this one complete.  Thanks again for your help.  Please feel free to close this issue.

chrismhinds, 2016-12-01:
I just realized that I made a backup of my Volumio image.  It's from October 2015, but it works!  I'm reverting back to it.  Thanks!

vassilmladenov, 2017-02-14:
Would you mind sharing said backup of your Volumio image?

closed 2018-03-26 by volumio