volumio/Volumio2 #899: Consume & Random Queue (2.032)

chsims1, 2016-12-03:
I have seen a number of posts on the forums about problems with random and consume playlists on earlier versions of Volumio, and I thought that they had been resolved.  Just noticing this evening a number of problems when random and consume are selected.

1.  Incorrect song title displayed on 'Playback' screen.

2.  Actually, possibly it is displays the last song played.  Also shown on the top of the 'Queue' screen ..... Playing "title"

3. Repeat of song that should have been consumed.

4. Sometimes it does work correctly.

5. Playlist stops being played.

6. Track stops unexpectedly. Playlist halted

Sorry if this report is rather messy, but I've added points as they occurred over about a 45 minute period.

Callenberg, 2016-12-04:
I am one of them that have reported here and there about these issues now and then. Lately I have been thinking more about the play buttons and play modes and playlists since they seem not to work that well. Some combinations do not work at all. 

Considering 1 & 2.
If the play mode is sequential (the default option) and the last track in the queue has been played and Volumio is finished playing (no repeat or consume) it marks the first track again as the next track to be played. Maybe that is what you have seen? 

Considering 3 & 4.
Yup, happens sometimes if Random + Consume modes are on, but not always as you have noticed. When that happens the player control buttons on the Player tab are not accessible (play, previous, next, time circle, . . .). Choosing another track brings the control back. This also might screw up the title (1 & 2 above). Bug.

Considering 5.
Happens to me if play mode is Repeat. Repeat seems not to work. The queue is not repeated. Playlists have some problems too. Bugs.

Considering 6. Sorry, can’t confirm that. Has never happened to me. Playing a track seems stable.

But what is right and was is wrong regarding playing tracks from the queue using different modes and combination of modes can always be discussed, but the implementation should behave consistently and fairly well fulfill general expectations.  I mean repeat should be repetitive (must be a bug though). 

Here is how I think it should be.

- Default the queue is a FiFo queue, first in, first out, when the last queued track has been played, the player stops.
- Repeat makes the FiFo queue circular. This means that Previous track and Next track works on a circular list of tracks.
- Random should be replaced by Shuffle, which means that the queue is randomly reorganized and the tracks are played only once.
- Random + Repeat means that the queue is re-shuffled after all the tracks have been played.
- Consume can be dropped.

  . . . but that is just what I think.
Actually I am fine with how Random is implemented in Volumio, but then some of the combinations are questionable. Random + Repeat make no sense and Repeat + Consume is contradictory. This is tricky.

Callenberg, 2016-12-04:
I for got to tell that I use Volumio 2.032 + Raspberry Pi 3 + IQaudIO P-DigiAMP+.

I also add some of my of earlier observations:
Reported in the Forum - [https://volumio.org/forum/volumio-031-play-mode-bugs-t5195.html](url)

**Repeat** doesn’t work at all if the tracks are added by using the Add to queue option in the Browser Tab. If the Play option is used it works actually once, but not twice. 
[Repeat should make the queue circular and repeat the queue in sequential order indefinitely.]

**Random** + **Consume** works like Consume only, no random function. Occasionally a track previously deleted is played. 
[Random + Consume is a combination that makes sense. Every track that is randomly played should be consumed (deleted).]

**Repeat** + **Consume**, it is the consume mode that wins. Questionable combination…  
[Repeat + Consume is contradictory. What is consumed cannot be repeated.]

**Random** + **Repeat** + Consume mostly this combination works only as consume, sometimes it plays a track that is not in the queue, or the player just gets instable. 
[Random + Repeat + Consume makes absolutely no sense.]

In addition . . the **Delete** option for individual tracks doesn’t work (X). The track keeps on playing and the player buttons are not working. 

Remarks:
Random in single mode works. Note that random mode is not a strict shuffle, the same track can be played twice. 
Consume in single mode works. 
Random + Repeat makes no sense since Random is implemented in such a way that it is playing the tracks indefinitely. 

The above was also reported here at the Github: [https://github.com/volumio/Volumio2/issues/868](url)

Playlists management doesn't work either, see: [https://volumio.org/forum/volumio-031-playlist-bugs-t5200.html](url)

chsims1, 2016-12-04:
Yes, it is all rather "muddy water."  I hadn't noticed the issues previously because I usually have a wide range of artists playing in a queue; in this case it was one artist and two albums so "irregularities" were really noticeable.

omehegan, 2017-01-01:
Repeat doesn't work in 2.041.

biva, 2017-03-15:
5 (Playlist stops being played) is happening for me on 2.118

closed 2017-06-02 by volumio