volumio/Volumio2 #901: v2.030, Tab MyMusic, difference between Update and Rescan

RastaX, 2016-12-04:
The difference between UPDATE and RESCAN is not clear to me, am i missing something obvious?

2lepus, 2016-12-04:
That's right. I also wondered what the difference might be. Maybe "rescan" deletes the library? But the result should be the same, iI suppose.

LotusGuy, 2016-12-14:
According to the MPD manual:

update:  Updates the music database: find new files, remove deleted files, update modified files.

rescan:  Same as update, but also rescans unmodified files.

Interestingly, MPD does support sending a URI to define a directory or file to perform the action on.  I did start looking at what it might take to update the UI to allow this to be selected, as an update on my library takes close to an hour.  But all the technologies used are new to me, so I'm not sure I'd make much headway.

Cheers.

xipmix, 2017-02-24:
Hi, I think this could be fixed up updating the tooltips in the UI.
```
diff --git a/src/app/i18n/locale-en.json b/src/app/i18n/locale-en.json
index 6b81051..ed7faaa 100644
--- a/src/app/i18n/locale-en.json
+++ b/src/app/i18n/locale-en.json
@@ -39,7 +39,7 @@
     "SONGS":"Tracks",
     "PLAYTIME":"Playtime",
     "TIME":"Time",
-    "RESCAN":"Rescan",
+    "RESCAN":"Remake music database entries for all files",
     "SYSTEM_VERSION":"System Version",
     "RELEASED":"Released",
     "CLOSE":"Close",
@@ -90,7 +90,7 @@
     "HIDE_DETAILS":"- Hide details",
     "INSTALL":"Install",
     "UNINSTALL":"Uninstall",
-    "UPDATE":"Update",
+    "UPDATE":"Update music database entries for any changed, new, or deleted files",
     "DETAILS":"Details",
     "DROP":"Click and select a plugin or drop it here.",
     "INSTALL_PROGRESS":"Installation Progress",
```
Perhaps you could open an issue there that points here?

closed 2017-06-02 by volumio