volumio/Volumio2 #904: Fix kernelsource

volumio, 2016-12-04:
https://volumio.org/forum/volumio-and-kernelsource-t5286.html#p24876

xipmix, 2016-12-05:
Is it not better to fix the sudoers file?  '/bin/systemctl' is in the NOPASSWD: list already so the 'echo' and the sudo -S seem unnecessary. Also the target scripts all have the right ```#!``` line so all that should be needed is ```chmod a+x``` ... but the execute bit is already set.

I prepared a patch but don't want to bother you with a PR if you are on it already.

volumio, 2016-12-05:
Please send, it will be useful!

xipmix, 2016-12-05:
I got a bit overexcited and gave kernelsource.sh a working over. See [kernelsource-part2](https://github.com/xipmix/Volumio2/tree/kernelsource-part2). It builds on the changes in the PR already sent.

volumio, 2016-12-05:
Ehehe, Ok I will test asap! thx!

xipmix, 2016-12-05:
I found some problems; calling pull.sh and kernelsource.sh needs sudo -S or changes to the sudoers file. The first can be handled by not requiring root permissions. The second needs [this](https://github.com/xipmix/Build/tree/kernelsource-part2)

closed 2017-11-30 by volumio