volumio/Volumio2 #905: Plugin deinstallation leaves config behind

jo-me, 2016-12-04:
After uninstalling a plugin, the config file(s) of the plugin under /data/configuration are not removed.

The method PluginManager.prototype.removePluginFromConfiguration() removes only the in-memory configuration of the plugin but not the files.

Aside from the residue this might cause sideeffects when reinstalling plugins .

jo-me, 2016-12-04:
Associated forum thread: https://volumio.org/forum/plugin-config-file-remains-after-uninstall-t5281.html

jo-me, 2016-12-04:
Another plugin deinstallation related issue is here https://github.com/volumio/Volumio2/issues/851

closed 2017-06-03 by volumio

macmpi, 2017-06-04:
Last I checked, it seemed plugin configuration folder was not removed upon uninstall.
Has this been fixed?

macmpi, 2018-02-14:
fixed with https://github.com/volumio/Volumio2/commit/e512b5064eff470f696d69612fb11e859d020d92: thanks!