volumio/Volumio2 #916: Airplay harware volume mixer control not possible - Volumio resets my changes in the config

suiluj, 2016-12-12:
Hello,

i installed Volumio and hardware volume control for my usb soundcard connected to my raspberry pi 3 is working for example for local file playback and web radio.

But for airplay it uses software volume which I don't want.

I already edited the config file of shairport-sync `/etc/shairport-sync.conf` and added the line `mixer_control_name = "Speaker";` in the alsa mixer part. This worked for me in the past when I used shairport-sync standalone.
But know it is not working. It seem that Volumio overwrite the shairport-sync config file on every startup and my edits are missing when i open the file again over ssh.

What can I do to prevent Volumio from changing back my custom configuration?

Thanks for your help.

volumio, 2016-12-12:
See here, you'll find the solution to your problem as well:
https://volumio.org/forum/airplay-password-disable-airplay-t719.html

suiluj, 2016-12-12:
Ah perfect! Thanks for this fast answer. That's exactly what I looked for!

I edited the template file as described in the forum post with `sudo nano  /volumio/app/plugins/music_service/airplay_emulation/shairport-sync.conf.tmpl`

I only had to change the line `output_device = "hw:1";` to `output_device = "hw:5";` because Volumio detects my USB soundcard on `hw:5`.

Now everything works great.
Thanks a lot for your help and this great software!

closed 2016-12-12 by suiluj