volumio/Volumio2 #919: MPD restarts randomly

pgporada, 2016-12-18:
```
Linux volumio 4.4.9-v7+ #884 SMP Fri May 6 17:28:59 BST 2016 armv7l GNU/Linux
Music Player Daemon 0.19.19
Volumio build released 12-12-2016
```

I have a 64GB exfat USB drive mounted via /etc/fstab. Every so often mpd will restart while browsing the library. The library will need to be rebuilt and mpd can restart several times during that process. I've never been able to have a fully built music library with this.

volumio, 2016-12-18:
Why did you mount it via fstab? 
To me it looks like it is conflicting with Volumio's automount, so remove your /etc/fstab mount first and let us know

pgporada, 2016-12-18:
The problem "feels" like it happens more when it is automounted. This is purely from a pragmatic standpoint. Additionally, songs will randomly stop playing. I can play the songs just fine through VLC or the cli on my machine. I'm not sure why that happens.

pgporada, 2016-12-18:
Here's an example of the error I get in `/var/log/volumio.log`
```

Dec 18 22:01:53 volumio mpd[550]: update: added USB/music/The Forthrights/Thanks!/01 Cryin'.mp3
Dec 18 22:01:53 volumio mpd[550]: update: added USB/music/The Forthrights/Thanks!/02 Terrible Wind (Demo).mp3
Dec 18 22:01:53 volumio mpd[550]: update: added USB/music/The Forthrights/Thanks!/03 Minds In Misery.mp3
Dec 18 22:01:53 volumio mpd[550]: update: added USB/music/The Forthrights/Thanks!/05 Like A Child (Demo).mp3
Dec 18 22:01:53 volumio mpd[550]: update: added USB/music/The Forthrights/Thanks!/06 Carla.mp3
Dec 18 22:01:53 volumio mpd[550]: update: added USB/music/The Forthrights/Thanks!/07 Policeman (Live In Chicago).mp3
Dec 18 22:01:53 volumio mpd[550]: update: added USB/music/The Forthrights/Thanks!/08 Jump   The Tide Is High.mp3
Dec 18 22:01:53 volumio mpd[550]: update: added USB/music/The Forthrights/Thanks!/09 Liquor Store Music.mp3
Dec 18 22:01:53 volumio systemd[1]: mpd.service: main process exited, code=killed, status=11/SEGV
Dec 18 22:01:53 volumio systemd[1]: Unit mpd.service entered failed state.
Dec 18 22:01:53 volumio systemd[1]: mpd.service holdoff time over, scheduling restart.
Dec 18 22:01:53 volumio systemd[1]: Stopping Music Player Daemon...
Dec 18 22:01:53 volumio systemd[1]: Starting Music Player Daemon...
Dec 18 22:01:53 volumio systemd[1]: Started Music Player Daemon.
Dec 18 22:01:54 volumio mpd[1199]: zeroconf: No global port, disabling zeroconf
Dec 18 22:01:54 volumio mount.exfat[288]: unknown entry type 0xfe
Dec 18 22:01:54 volumio mount.exfat[288]: failed to open directory '/Vyvyan'
Dec 18 22:01:55 volumio mount.exfat[288]: unknown entry type 0xfe
Dec 18 22:01:55 volumio mount.exfat[288]: failed to open directory '/Soul Radics/Big Shot'
Dec 18 22:01:56 volumio mount.exfat[288]: unknown entry type 0xfe
Dec 18 22:01:56 volumio mount.exfat[288]: failed to open directory '/ODESZA'
Dec 18 22:01:57 volumio mount.exfat[288]: unknown entry type 0x8d
Dec 18 22:01:57 volumio mount.exfat[288]: failed to open directory '/Milo'
Dec 18 22:01:57 volumio mount.exfat[288]: unknown entry type 0xe1
Dec 18 22:01:57 volumio mount.exfat[288]: failed to open directory '/Chumped'
Dec 18 22:01:57 volumio mpd[1199]: client: [0] opened from ::ffff:127.0.0.1:39024
Dec 18 22:02:00 volumio volumio[785]: info: CALLMETHOD: audio_interface upnp stopUpnpPlayback [object Object]
Dec 18 22:02:00 volumio volumio[785]: info: CoreCommandRouter::executeOnPlugin: upnp , stopUpnpPlayback
Dec 18 22:02:00 volumio volumio[785]: info: Stopping playback through UPNP
Dec 18 22:02:00 volumio volumio[785]: info: [1482098520490] CoreStateMachine::setConsumeUpdateService undefined
```

volumio, 2016-12-18:
Try to inspect your thumb drive for errors, as it seems most likely the case here

volumio, 2017-06-03:
``

closed 2017-06-03 by volumio