volumio/Volumio2 #926: Reconnecting to WiFi network

Bergmann1, 2016-12-22:
Hello,

I have an issue with reconnecting Volumio to Wifi network. Sometimes can happen that the WiFi router reboot and the WiFi network goes down. After the restart of the router and after the WiFi network goes up again, Volumio does not reconnect automatically to the network, but I have to reboot the RP since I am accessing through the web interface. Does anyone had the similar issue and how did it solve it? With previous version of Volumio it took sometime to reconnect automatically, but it reconnected at the end.

I am using RP 2 with the latest Volumio2 update. The network card is EDIMAX EW-7811UN.

Thank you and regards.

lgnap, 2017-01-01:
Yes, I have same symptoms.

Bergmann1, 2017-01-01:
Temporarily I got back to 1.55 version, I found it far more stable and reliable despite I very like version 2.x

balbuze, 2017-01-01:
Yes I've got the same problem. I power off my router and on every morning. And Volumio can't reconnect the network.

balbuze, 2017-01-06:
I wrote a plugin that reconnect wifi. Please have look here  to test : https://github.com/balbuze/volumio-plugins/tree/master/plugins/system_controller/wifireconnect
It may not work as expected....Let me know

balbuze, 2017-01-06:
But I have tested it on two device and for me it works !

xipmix, 2017-01-07:
I wonder if would be better to test just local connectivity - ie can you get to the local router, than going to an external address; this makes it easier to tell if problems are with volumio or the local router/wifi access point.```ifconfig <wlan interface>``` and ```netstat -rn``` would give enough information I think.

xipmix, 2017-01-07:
Actually the output of ```ip address show <device>``` and ```ip route show``` might be easier to parse. I don't know if this is part of the default volumio install though and it might be too much bloat.

balbuze, 2017-01-07:
I agree with you for local IP. But I can't guess IP for each case. It's because I set as default IP Google one. But you can set the IP you want in the plugin setting. But I have a problem in some cases. It seems to conflict with hotspot. I have to investigate..

balbuze, 2017-01-10:
I've done a lot of test. The script used in the plugin works for a short network drop. But remains for a long network down. It seems due to dhcp that not is not renewed. When I stop my router, wait more than one hours and restart, Volumio can't reconnect even when restarting interface. Only a reboot do the job. and it seems something is wrong with dhcp. I use a dynamic address. I'm not a network specialist.... If someone knows....

volumio, 2017-01-11:
In fact, this is done by design.
If network drops, volumio attempts to reconnect for 30 seconds, then starts the hotspot.
So, if your network is not available after 30 seconds, it will start hotspot mode. We can add an option not to recreate hotspot if wireless fails, but to retry wireless connection

chsims1, 2017-01-11:
Or perhaps an option to alter the reconnection time?? 30s seems very short to me, but ofc it depends on an individual's circumstances/needs.

macmpi, 2017-01-11:
> In fact, this is done by design.
> If network drops, volumio attempts to reconnect for 30 seconds, then starts the hotspot.

What if Volumio simply does not start hotspot if user did set Hotspot to OFF and saved settings as such?
Trying to restart hotspot if user specifically asked to turn it OFF seems overruling user's preference.
In the described case, obvious user's expectation is that Volumio will just keep trying connecting to Wifi AP.

In general Volumio should probably keep doing what user did specifically setup within Preferences, and not intervene with different initiatives which may be very confusing for users that are not "in the know".

Makes sense?

balbuze, 2017-01-11:
I set that by default  in /etc/dhcp/dhcpd.conf
default-lease-time 600;
max-lease-time 3600;
by default, no value, that means 86400 sec.
I have to see if it helps, combine with interface restarts (done by plugin)

balbuze, 2017-01-13:
Now my device is still connected for two days even with the wifi router off all the night :-)

macmpi, 2017-01-13:
Interesting workaround.
However this discussion shows Hotspot OFF actually means **Hostpot AUTO** (automatic mode that kicks-in if something goes wrong).
How about then labeling it as such, and add a true **Hotspot OFF** for user really willing to turn it off, and have Volumio keep trying to connect to AP.

User may want Hotspot:
- OFF, no matter what (typically rely on home AP)
- ON, no matter what (typically no home AP, or peer to peer)
- AUTO: fallback as described if AP connection fails, and initial setup at boot

volumio, 2017-01-13:
That's clever! I will implement it this way

macmpi, 2017-01-13:
Cool! Still considering [create_ap](https://github.com/oblique/create_ap)?

earlchew, 2017-02-04:
I did some research to see if it would be possible to run both HotSpot and AccessPoint simultaneously. This would be a great solution since each wireless network would run independently. Unfortunately, the ability to run hostapd and wpa_supplicant depends critically on the feature supported by the hardware and driver: http://www.0xf8.org/2016/02/using-your-raspberry-pi-zeros-usb-wifi-adapter-as-both-wifi-client-and-access-point/. This means that this solution cannot be used in the general case.

volumio, 2017-02-04:
I would say that your original plan might be the best option...

jsun-g, 2017-02-09:
I am experiencing this issue as well (RPi3 w/Volumio 2).
Since I am deploying RPi in a car, I want the Volumio Hotspot off as a matter of security.
I want the ability of Volumio to connect to my home's Wifi AP after long periods of no Wifi coverage (for e.g. when I park the car in the garage after a long day away from home, to download additional music from home laptop via automated script).

Eagerly awaiting a solution to the Wifi restoring.

earlchew, 2017-02-11:
As far as I can tell, the current volumio code does not restart the hotspot if the setting is **OFF**:
```javascript
               if (hotspotjson != undefined
                  && hotspotjson.enable_hotspot != undefined
                  && hotspotjson.enable_hotspot.value != undefined
                  && !hotspotjson.enable_hotspot.value) {
                                        console.log('Hotspot is disabled, not starting it');
                                    launch(ifconfigWlan, "configwlanup", true, function(err) {
                                        console.log("ifconfig " + err);
                                        });
                } else {

                launch(ifconfigHotspot, "confighotspot", true, function(err) {
                        console.log("ifconfig " + err);
                        launch(starthostapd,"hotspot" , false, function() {
                                wstatus("hotspot");
                        });
                });
```

earlchew, 2017-02-15:
I've written a description of [work in progress](https://github.com/volumio/Build/issues/143#issuecomment-279920639).

fr1sty, 2017-05-28:
My wifi is off during the nights. So i have to restart my volumio every day to make it reconnect. I tried the plugin https://github.com/balbuze/volumio-plugins/tree/master/plugins/system_controller/wifireconnect and i disabled the hotspot mode but there's no difference. Is there another way to make wifi reconnect automatically?

bvp663, 2017-06-13:
Same issue here. I put the script I found [here](http://weworkweplay.com/play/rebooting-the-raspberry-pi-when-it-loses-wireless-connection-wifi/) onto the Raspberry Pi but can't seem to bring up the `crontab` in order to have it execute regularly.

biva, 2017-09-06:
@bvp663 see https://github.com/balbuze/volumio-plugins/issues/64#issuecomment-327517272

XavierDupessey, 2017-11-12:
Hi!

My wifi is scheduled to turn off during work and sleep time. I didn't want to test the entire Internet connection but only the wifi state because I still wanted to be able to use AirPlay on my local network.

So I use the following script for that purpose and it works well on my rpi3 (even after several hours of wifi off):

```
sudo mkdir /usr/local/bin/wifi-check 
cd /usr/local/bin/wifi-check 
sudo nano wifi-check.sh 
    #! /bin/sh 
 
    ssid=$(/sbin/iwgetid --raw) 
 
    if [ -z "$ssid" ] 
    then 
        echo "Wifi is down, reconnecting..." 
        /sbin/ifconfig wlan0 down 
        sleep 5 
        systemctl restart wireless 
    fi 
 
    echo "wifi-check done" 
 
sudo chmod +x /usr/local/bin/wifi-check/wifi-check.sh 
 
sudo crontab -e 
    */5 * * * * /usr/local/bin/wifi-check/wifi-check.sh 
```

Hope it helps...

volumio, 2018-03-26:
Added

closed 2018-03-26 by volumio