volumio/Volumio2 #930: Can't connect to ssh since last update

youpilai, 2016-12-23:
I just updated to the latest dev volumio (2.046) and can't play anymore music. I first thought it was cause by my own modifications, so i've done a factory reset.
It reversed the system to 2.001, so i've done an update to 2.041, and can't be able to hear music through HifiBerry DAC output. (like #927 ?) 
I so activated the Test Mode, through the volumio.local/dev page, and updated one more time to 2.046.
Now i can't even add a song to play. Anytime i want to play a song, add a song to the queue, clear and play etc... the player restart (wheel thing followed by startup sound)
Also i can't anymore connect via ssh : the terminal is opening (good IP) with putty, i can enter the login, i can enter the password but every time it closes the connection ("Server unexpectedly closed network connection")
I sadly didn't keep the last working version (for me), so i can't reinstall it.
I didn't recall exactly the version, but it was something like 2.037 from december 7th or so

gkkpch, 2016-12-27:
Yes, this is a know issue that appeared just before the holidays. There is an issue with http communication, @volumio expects this to be solved by the raspbian community but until he comes back there will be no update
[2016-12-08-volumio-2.038](http://updates.volumio.org/pi/volumio/2.038/volumio-2.038-2016-12-08-pi.img.zip)
[2016-12-07-volumio-2.037](http://updates.volumio.org/pi/volumio/2.037/volumio-2.037-2016-12-07-pi.img.zip)
[2016-12-06-volumio-2.036](http://updates.volumio.org/pi/volumio/2.036/volumio-2.036-2016-12-06-pi.img.zip)

youpilai, 2016-12-28:
Thanks gkkpch for the links
I'll try to make a clean install from one of these in the meantime

macmpi, 2017-01-23:
Is that now fixed by some interim releases since 2.041? (like 2.056 or else?).

Been hit on rpi0 + pHat DAC: all is fine with HDMI sound, but device is crawling to death as soon a i2S is set with DAC.

volumio, 2017-06-13:
Duplicate of https://github.com/volumio/Volumio2/issues/599

closed 2017-06-13 by volumio